zephyrproject-rtos/zephyr #18365: [Coverity CID :203434]Control flow issues in /drivers/timer/xlnx_psttc_timer.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /drivers/timer/xlnx_psttc_timer.c
 Category: Control flow issues
 Function: xttc_calculate_interval
 Component: Drivers
 CID: 203434
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-08-18:
@bbolen this code came from your PR #9316 from this commit 4ef9d4b , could you take a look at it?

[image]

bbolen, 2019-08-19:
Yes, I'll take a look.

galak, 2019-08-19:
@bbolen looks like not a bug, but will let you look in more detail.

LeiW000, 2019-08-21:
We should ignore it in coverity. It is possible that "TIMER_INPUT_CLKHZ" is defined with a very large number by user in qemu_cortex_r5.dts.

I have updated "CID 203434" in coverity to ignore it. So, I think we can close the issue now.

closed 2019-08-21 by LeiW000