zephyrproject-rtos/zephyr #18373: [Coverity CID :203399]API usage errors in /samples/boards/olimex_stm32_e407/ccm/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /samples/boards/olimex_stm32_e407/ccm/src/main.c
 Category: API usage errors
 Function: print_var_values
 Component: Samples
 CID: 203399
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-08-18:
@erwango: Would you be able to take a look at this?

erwango, 2019-08-27:
@aescolar looking at it. One question though: how come this has not been seen earlier?

erwango, 2019-08-27:
@galak , since 2017 ?

aescolar, 2019-08-30:
> One question though: how come this has not been seen earlier?

@erwango I do not now (I just helped triage these issues. I have a very limited insight into how we use Coverity). But maybe more files are now included in the analysis, or maybe Coverity detects now more issues than before.
@aasthagr may know why this (and many other old issues) suddenly appeared?

lowlander, 2019-09-13:
Do I understand correct that it only complains about the %d needing to be %ld ?

Is there any special thing I have to write in the commit messages when making a PR for this ?

lowlander, 2019-09-13:
OK I just compiled that example and the compiler does not give a warning, when I change the %d into %ld it does give a warning. 

So what should I do?

erwango, 2019-09-16:
@aasthagr, can you help here ? Does coverity use specific compile options?

PR #19258 (2019-09-19): samples: olimex_stm32_e407: fix coverity warning
Fix a Coverity warning when using %d to print the result
of a sizeof(...)

Since the result of the sizeof() use in the example
will always fit in a int, simply cast the result
to a int so the %d is always correct.

Fixes: #18373
commit a720d38f (2019-09-19): samples: olimex_stm32_e407: fix coverity warning

Fix a Coverity warning when using %d to print the result
of a sizeof(...)

Since the result of the sizeof() use in the example
will always fit in a int, simply cast the result
to a int so the %d is always correct.

Fixes: #18373
closed 2019-09-19 by galak