zephyrproject-rtos/zephyr #18374: [Coverity CID :203523]Error handling issues in /samples/drivers/counter/alarm/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /samples/drivers/counter/alarm/src/main.c
 Category: Error handling issues
 Function: test_counter_interrupt_fn
 Component: Samples
 CID: 203523
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203523:  Error handling issues  (CHECKED_RETURN)
/samples/drivers/counter/alarm/src/main.c: 37 in test_counter_interrupt_fn()
31     	config->ticks = config->ticks * 2U;
32     
33     	printk("Set alarm in %u sec (%u ticks)\n",
34     	       (u32_t)(counter_ticks_to_us(counter_dev,
35     					   config->ticks) / USEC_PER_SEC),
36     	       config->ticks);
>>>     CID 203523:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "counter_set_channel_alarm" without checking return value (as is done elsewhere 6 out of 7 times).
37     	counter_set_channel_alarm(counter_dev, ALARM_CHANNEL_ID, user_data);
38     }
39     
40     void main(void)
41     {
42     	struct device *counter_dev;
```

aescolar, 2019-08-18:
from da744314ca4 #12668

commit e15852cd (2019-08-28): samples: counter: Add err check on set_alarm call

Error checking was missing following call to
counter_set_channel_alarm.
This was reported in coverity report 203523.

Fixes #18374

Additionally remove unneeded err initialization in main function