zephyrproject-rtos/zephyr #18378: [Coverity CID :203537]Error handling issues in /samples/subsys/nvs/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /samples/subsys/nvs/src/main.c
 Category: Error handling issues
 Function: main
 Component: Samples
 CID: 203537
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-08-18:
from b1e45b413ef #6391

jenmwms, 2020-02-06:
This looks 'Fixed' in the report page (link below). Can someone confirm? @aasthagr 
https://scan9.coverity.com/reports.htm#v39298/p12996/fileInstanceId=55470800&defectInstanceId=6797685&mergedDefectId=203537

Laczen, 2020-02-07:
Hi @jenmwms, this is not really an issue. However if you would like to fix it completely all calls of nvs_write() or nvs_delete() that are not using the return value can be changed to `(void)nvs_write` and `(void)nvs_delete`.

jenmwms, 2020-02-07:
Thanks @Laczen!

jenmwms, 2020-02-25:
I contemplated a patch that checked the return values, but it is bulky and checking the return in that context does not appear to add value. So it is cleaner to simply change those few calls to (void).

PR #23060 (2020-02-27): samples: subsys: nvs: unchecked return values CID#203537
The sample contained calls that were not using the return
value, which was detected by Coverity Scan as an issue. This
commit fixes it by changing to (void).

Fixes #18378
CID#203537
commit d6373a38 (2020-02-27): samples: subsys: nvs: unchecked return values CID#203537

The sample contained calls that were not using the return
value, which was detected by Covery Scan as an issue. This
commit fixes it by changing to (void).

Fixes #18378
CID#203537
closed 2020-02-27 by jhedberg