zephyrproject-rtos/zephyr #18380: [Coverity CID :203461]Null pointer dereferences in /subsys/bluetooth/controller/ll_sw/ull.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/bluetooth/controller/ll_sw/ull.c
 Category: Null pointer dereferences
 Function: rx_demux_rx
 Component: Bluetooth
 CID: 203461
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

ioannisg, 2019-08-19:
```
1/**
1452 * @brief Dispatch rx objects
1453 * @details Rx objects are only peeked, not dequeued yet.
1454 *   Execution context: ULL high priority Mayfly
1455 */
1456static inline int rx_demux_rx(memq_link_t *link, struct node_rx_hdr *rx)
1457{
1458        /* Demux Rx objects */
    deref_ptr: Directly dereferencing pointer rx.
1459        switch (rx->type) {
1460        case NODE_RX_TYPE_EVENT_DONE:
1461        {
1462                memq_dequeue(memq_ull_rx.tail, &memq_ull_rx.head, NULL);
1463                rx_demux_event_done(link, rx);
1464        }
1465        break;
1466
1467#if defined(CONFIG_BT_OBSERVER) || \
1468        defined(CONFIG_BT_CTLR_SCAN_REQ_NOTIFY) || \
1469        defined(CONFIG_BT_CTLR_PROFILE_ISR) || \
1470        defined(CONFIG_BT_CTLR_ADV_INDICATION) || \
1471        defined(CONFIG_BT_CTLR_SCAN_INDICATION)
1472#if defined(CONFIG_BT_OBSERVER)
1473        case NODE_RX_TYPE_REPORT:
1474#endif /* CONFIG_BT_OBSERVER */
1475
1476#if defined(CONFIG_BT_CTLR_ADV_EXT)
1477        case NODE_RX_TYPE_EXT_1M_REPORT:
1478        case NODE_RX_TYPE_EXT_CODED_REPORT:
1479#endif /* CONFIG_BT_CTLR_ADV_EXT */
1480
1481#if defined(CONFIG_BT_CTLR_SCAN_REQ_NOTIFY)
1482        case NODE_RX_TYPE_SCAN_REQ:
1483#endif /* CONFIG_BT_CTLR_SCAN_REQ_NOTIFY */
1484
1485#if defined(CONFIG_BT_CTLR_PROFILE_ISR)
1486        /* fallthrough */
1487        case NODE_RX_TYPE_PROFILE:
1488#endif /* CONFIG_BT_CTLR_PROFILE_ISR */
1489
1490#if defined(CONFIG_BT_CTLR_ADV_INDICATION)
1491        case NODE_RX_TYPE_ADV_INDICATION:
1492#endif /* CONFIG_BT_CTLR_ADV_INDICATION */
1493
1494#if defined(CONFIG_BT_CTLR_SCAN_INDICATION)
1495        case NODE_RX_TYPE_SCAN_INDICATION:
1496#endif /* CONFIG_BT_CTLR_SCAN_INDICATION */
1497        {
1498                memq_dequeue(memq_ull_rx.tail, &memq_ull_rx.head, NULL);
1499                ll_rx_put(link, rx);
1500                ll_rx_sched();
1501        }
1502        break;
1503#endif /* CONFIG_BT_OBSERVER ||
1504        * CONFIG_BT_CTLR_SCAN_REQ_NOTIFY ||
1505        * CONFIG_BT_CTLR_PROFILE_ISR ||
1506        * CONFIG_BT_CTLR_ADV_INDICATION ||
1507        * CONFIG_BT_CTLR_SCAN_INDICATION
1508        */
1509
1510#if defined(CONFIG_BT_CONN)
1511        /* fallthrough */
1512        case NODE_RX_TYPE_CONNECTION:
1513        {
1514                memq_dequeue(memq_ull_rx.tail, &memq_ull_rx.head, NULL);
1515                ull_conn_setup(link, rx);
1516        }
1517        break;
1518
1519        case NODE_RX_TYPE_DC_PDU:
1520        {
1521                int nack;
1522
1523                nack = ull_conn_rx(link, (void *)&rx);
1524                if (nack) {
1525                        return nack;
1526                }
1527
1528                memq_dequeue(memq_ull_rx.tail, &memq_ull_rx.head, NULL);
1529
    CID 203461 (#1 of 1): Dereference before null check (REVERSE_INULL)check_after_deref: Null-checking rx suggests that it may be null, but it has already been dereferenced on all paths leading to the check.
1530                if (rx) {
1531                        ll_rx_put(link, rx);
1532                        ll_rx_sched();
1533                }
1534        }
1535        break;
1536#endif /* CONFIG_BT_CONN */
1537
1538        default:
1539        {
1540#if defined(CONFIG_BT_CTLR_USER_EXT)
1541                /* Try proprietary demuxing */
1542                rx_demux_rx_proprietary(link, rx, memq_ull_rx.tail,
1543                                        &memq_ull_rx.head);
1544#else
1545                LL_ASSERT(0);
1546#endif /* CONFIG_BT_CTLR_USER_EXT */
1547        }
1548        break;
1549        }
1550
1551        return 0;
1552}
```

ioannisg, 2019-08-19:
@cvinayak ^^

cvinayak, 2019-08-19:
rx is in-out parameter in line 1523. Here it can return NULL for rx. But at 1459, it will not be NULL, as the caller will pass non-NULL rx pointer.

This check at line 1530 is intentional.

carlescufi, 2019-08-19:
Intentional, dismissed in Coverity

closed 2019-08-19 by carlescufi