zephyrproject-rtos/zephyr #18382: [Coverity CID :203524]Incorrect expression in /subsys/bluetooth/controller/ticker/ticker.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/bluetooth/controller/ticker/ticker.c
 Category: Incorrect expression
 Function: ticker_enqueue
 Component: Bluetooth
 CID: 203524
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203524:  Incorrect expression  (UNUSED_VALUE)
/subsys/bluetooth/controller/ticker/ticker.c: 375 in ticker_enqueue()
369     
370     	node = &instance->nodes[0];
371     	ticker_new = &node[id];
372     	ticks_to_expire = ticker_new->ticks_to_expire;
373     
374     	current = instance->ticker_id_head;
>>>     CID 203524:  Incorrect expression  (UNUSED_VALUE)
>>>     Assigning value from "current" to "previous" here, but that stored value is overwritten before it can be used.
375     	previous = current;
376     
377     	/* Find insertion point for new ticker node and adjust ticks_to_expire
378     	 * relative to insertion point
379     	 */
380     	previous = TICKER_NULL;
```

aescolar, 2019-08-18:
@mtpr-ot introduced in 3a9173afe15

mtpr-ot, 2019-08-19:
This is a copy-paste mistake when doing the CONFIG_BT_TICKER_COMPATIBILITY_MODE conditions. It has no impact as the desired initialization is in line 380. Will make a PR with an update.

mtpr-ot, 2019-08-19:
Solution merged

closed 2019-08-19 by mtpr-ot