zephyrproject-rtos/zephyr #18386: [Coverity CID :203443]Memory - corruptions in /subsys/bluetooth/host/rfcomm.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/bluetooth/host/rfcomm.c
 Category: Memory - corruptions
 Function: bt_rfcomm_dlc_send
 Component: Bluetooth
 CID: 203443
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

jhedberg, 2019-08-18:
There are several things that look messed up with this code. I don't have time to go digging through the RFCOMM spec to figure out what exactly is the right thing. Maybe this is the final straw and we should just remove BR/EDR support from the tree. Downgrading to low priority however since this is considered experimental code.

joerchan, 2020-01-03:
@aasthagr Could you possibly check if this issue has been resolved with the fix that was recently submitted to RFCOMM?

joerchan, 2020-01-21:
> aasthagr Could you possibly check if this issue has been resolved with the fix that was recently submitted to RFCOMM?

Answering my own question. No it does not.

nashif, 2020-03-11:
this is marked fixed in coverity with comment

"Source code is not available because it has been purged."

nashif, 2021-03-05:
the issue is still there in coverity

joerchan, 2021-03-05:
This is intentional, rfcomm has a variable length length field, which is confusing when we cast the uint8_t field to an uint16_t field instead, but the extra byte has been pushed to the net_buf.

carlescufi, 2021-05-20:
@MaureenHelm assigning this to you since @MarkWangChinese is not yet part of the org, and this is BR/EDR with which you've shown an interest in keeping alive.

MaureenHelm, 2021-05-24:
@MarkWangChinese can you take a look this week?

MarkWangChinese, 2021-05-27:
I created one PR https://github.com/zephyrproject-rtos/zephyr/pull/35706

PR #35706 (2021-06-24): bluetooth: rfcomm: fix the coverity
- the coverity CID is 203443 memory corruptions

Fixes: #18386
closed 2021-06-24 by jhedberg