zephyrproject-rtos/zephyr #18391: [Coverity CID :203403]Memory - corruptions in /subsys/disk/disk_access_usdhc.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/disk/disk_access_usdhc.c
 Category: Memory - corruptions
 Function: usdhc_cmd_rsp
 Component: Other
 CID: 203403
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-08-18:
@JunYangNXP Could you please look at this?

JunYangNXP, 2019-08-21:
> @JunYangNXP Could you please look at this?
> (Maureen, I assign you as Jun is not a project collaborator)

Hi, how can I see the issue description? @MaureenHelm  @aescolar

aescolar, 2019-08-21:
@JunYangNXP  Here you have:
```static int usdhc_cmd_rsp(struct usdhc_priv *priv)
1079{
1080        u32_t i;
1081        USDHC_Type *base = priv->host_config.base;
1082        struct usdhc_cmd *cmd = &priv->op_context.cmd;
1083
    	1. Condition cmd->rsp_type != SDHC_RSP_TYPE_NONE, taking true branch.
1084        if (cmd->rsp_type != SDHC_RSP_TYPE_NONE) {
1085                cmd->response[0U] = base->CMD_RSP0;
    	2. Condition cmd->rsp_type == SDHC_RSP_TYPE_R2, taking true branch.
1086                if (cmd->rsp_type == SDHC_RSP_TYPE_R2) {
1087                        cmd->response[1U] = base->CMD_RSP1;
1088                        cmd->response[2U] = base->CMD_RSP2;
1089                        cmd->response[3U] = base->CMD_RSP3;
1090
1091                        i = 4U;
1092                        /* R3-R2-R1-R0(lowest 8 bit is invalid bit)
1093                         * has the same format
1094                         * as R2 format in SD specification document
1095                         * after removed internal CRC7 and end bit.
1096                         */
1097                        do {
    	
CID 203403 (#1 of 1): Out-of-bounds write (OVERRUN)
9. overrun-local: Overrunning array cmd->response of 4 4-byte elements at element index 4294967295 (byte offset 17179869183) using index i - 1U (which evaluates to 4294967295).
1098                                cmd->response[i - 1U] <<= 8U;
    	3. Condition i > 1U, taking true branch.
    	5. Condition i > 1U, taking false branch.
    	6. cond_at_most: Checking i > 1U implies that i may be up to 1 on the false branch.
1099                                if (i > 1U) {
1100                                        cmd->response[i - 1U] |=
1101                                                ((cmd->response[i - 2U] &
1102                                                0xFF000000U) >> 24U);
1103                                }
    	4. Condition i--, taking true branch.
    	7. decr: Decrementing i. The value of i is now 0.
    	8. Condition i--, taking true branch.
1104                        } while (i--);
1105                }
1106        }
1107        /* check response error flag */
1108        if ((cmd->rsp_err_flags) &&
1109                ((cmd->rsp_type == SDHC_RSP_TYPE_R1) ||
1110                (cmd->rsp_type == SDHC_RSP_TYPE_R1b) ||
1111                (cmd->rsp_type == SDHC_RSP_TYPE_R6) ||
1112                (cmd->rsp_type == SDHC_RSP_TYPE_R5))) {
1113                if (((cmd->rsp_err_flags) & (cmd->response[0U])))
1114                        return -EIO;
1115        }
1116
1117        return 0;
1118}
```

PR #18786 (2019-08-30): disk: Fix usdhc driver out-of-bounds array access
The usdhc driver was incorrectly post-decrementing a do/while loop
iterator, causing an extra iteration of the loop and an out-of-bounds
array access. Change the iterator to pre-decrement instead.

Tested samples/subsys/fs/fat_fs on the mimxrt1050_evk board.

Coverity-CID: 203403

Fixes #18391

@JunYangNXP please review
closed 2019-08-30 by ioannisg