zephyrproject-rtos/zephyr #18393: [Coverity CID :203541]Integer handling issues in /subsys/net/ip/6lo.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/net/ip/6lo.c
 Category: Integer handling issues
 Function: uncompress_nh_udp
 Component: Networking
 CID: 203541
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

alexanderwachter, 2019-08-18:
Coverty:
```
Line 1271: udp->dst_port = htons((NET_6LO_NHC_UDP_4_BIT_PORT << 4) | (*cursor & 0x0F));
result_independent_of_operands: ((61616 /* 0xf0b << 4 */) | (*cursor & 0xf)) >> 8 is 0xf0 regardless of the values of its operands. This occurs as the bitwise first operand of "&".
```

This issue occurs because of the htons in case of a byte swap. IMO this code is correct.

jukkar, 2019-08-27:
False positive.

closed 2019-08-27 by jukkar