zephyrproject-rtos/zephyr #18397: [Coverity CID :203512]Memory - corruptions in /subsys/net/l2/ppp/ipv6cp.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/net/l2/ppp/ipv6cp.c
 Category: Memory - corruptions
 Function: ipv6cp_up
 Component: Networking
 CID: 203512
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203512:  Memory - corruptions  (OVERRUN)
/subsys/net/l2/ppp/ipv6cp.c: 448 in ipv6cp_up()
442     	peer_lladdr.addr = ctx->ipv6cp.peer_options.iid;
443     	peer_lladdr.len = sizeof(ctx->ipv6cp.peer_options.iid);
444     
445     	/* TODO: What should be the type? */
446     	peer_lladdr.type = NET_LINK_DUMMY;
447     
>>>     CID 203512:  Memory - corruptions  (OVERRUN)
>>>     Overrunning callee's array of size 6 by passing argument "peer_lladdr.len" (which evaluates to 8) in call to "net_ipv6_nbr_add".
448     	nbr = net_ipv6_nbr_add(ctx->iface, &peer_addr, &peer_lladdr,
449     			       false, NET_IPV6_NBR_STATE_STATIC);
450     	if (!nbr) {
451     		NET_ERR("[%s/%p] Cannot add peer %s to nbr table",
452     			fsm->name, fsm,
453     			log_strdup(net_sprint_addr(AF_INET6,
```

jukkar, 2019-08-27:
The Coverity comment "Indexing array of size 6 with lladdr->len minus an offset in call to memcmp. [Note: The source code implementation of the function has been overridden by a builtin model.]" is wrong. The code allocates 8 bytes for the link address. Verified this by compiling and printing the values.

closed 2019-08-27 by jukkar