zephyrproject-rtos/zephyr #18406: [Coverity CID :203511]Memory - corruptions in /subsys/net/lib/lwm2m/ipso_buzzer.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/net/lib/lwm2m/ipso_buzzer.c
 Category: Memory - corruptions
 Function: buzzer_create
 Component: Networking
 CID: 203511
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203511:  Memory - corruptions  (OVERRUN)
/subsys/net/lib/lwm2m/ipso_buzzer.c: 212 in buzzer_create()
206     	buzzer_data[avail].level.val1 = 50; /* 50% */
207     	buzzer_data[avail].delay_duration.val1 = 1; /* 1 seconds */
208     	buzzer_data[avail].obj_inst_id = obj_inst_id;
209     
210     	(void)memset(res[avail], 0,
211     		     sizeof(res[avail][0]) * ARRAY_SIZE(res[avail]));
>>>     CID 203511:  Memory - corruptions  (OVERRUN)
>>>     Calling "init_res_instance" with "res_inst[avail]" and "6U" is suspicious because the function call may access "res_inst" at byte "avail * sizeof (struct lwm2m_engine_res_inst [6]) + 71UL".
212     	init_res_instance(res_inst[avail], ARRAY_SIZE(res_inst[avail]));
213     
214     	/* initialize instance resource data */
215     	INIT_OBJ_RES(BUZZER_ON_OFF_ID, res[avail], i,
216     		     res_inst[avail], j, 1, true,
217     		     &buzzer_data[avail].onoff,
```

jukkar, 2019-08-27:
False positive

closed 2019-08-27 by jukkar