zephyrproject-rtos/zephyr #18407: [Coverity CID :203519]Memory - corruptions in /subsys/net/lib/lwm2m/ipso_light_control.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/net/lib/lwm2m/ipso_light_control.c
 Category: Memory - corruptions
 Function: light_control_create
 Component: Networking
 CID: 203519
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203519:  Memory - corruptions  (OVERRUN)
/subsys/net/lib/lwm2m/ipso_light_control.c: 166 in light_control_create()
160     	power_factor_value[avail].val2 = 0;
161     	colour[avail][0] = '\0';
162     	units[avail][0] = '\0';
163     
164     	(void)memset(res[avail], 0,
165     		     sizeof(res[avail][0]) * ARRAY_SIZE(res[avail]));
>>>     CID 203519:  Memory - corruptions  (OVERRUN)
>>>     Calling "init_res_instance" with "res_inst[avail]" and "8U" is suspicious because the function call may access "res_inst" at byte "avail * sizeof (struct lwm2m_engine_res_inst [8]) + 95UL".
166     	init_res_instance(res_inst[avail], ARRAY_SIZE(res_inst[avail]));
167     
168     	/* initialize instance resource data */
169     	INIT_OBJ_RES_DATA(LIGHT_ON_OFF_ID, res[avail], i, res_inst[avail], j,
170     			  &on_off_value[avail], sizeof(*on_off_value));
171     	INIT_OBJ_RES_DATA(LIGHT_DIMMER_ID, res[avail], i, res_inst[avail], j,
```

jukkar, 2019-08-27:
False positive

closed 2019-08-27 by jukkar