zephyrproject-rtos/zephyr #18417: [Coverity CID :203501]Memory - corruptions in /subsys/net/lib/lwm2m/ipso_timer.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /subsys/net/lib/lwm2m/ipso_timer.c
 Category: Memory - corruptions
 Function: timer_create
 Component: Networking
 CID: 203501
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

mike-scott, 2019-08-18:
Lines of code in question:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/lwm2m/ipso_timer.c#L334
```
	(void)memset(res[avail], 0,
		     sizeof(res[avail][0]) * ARRAY_SIZE(res[avail]));
```
Description of issue is:
```
CID 203501 (#1 of 1): Out-of-bounds access (OVERRUN)
11. overrun-buffer-arg: Calling memset with res[avail] and 264U is suspicious because the function call may access res at byte avail * sizeof (struct lwm2m_engine_res [11]) + 263UL. [Note: The source code implementation of the function has been overridden by a builtin model.]
```

Maybe, I'm misunderstanding the description of the issue, but for any actual value of avail (the object instance index), we **should** be able to safely clear out the sub array of 11 resource records (each 24 bytes in size for a total of 264).

jukkar, 2019-08-27:
I did a small test and printed the actual values:
```
sizeof(res[avail][0])  = 24
ARRAY_SIZE(res[avail]) = 11
sizeof(res[avail][0]) * ARRAY_SIZE(res[avail]) = 264
sizeof(res[avail]) = 264
```

So you could just say `(void)memset(res[avail], 0, sizeof(res[avail])` here as that is a bit more readable.

Anyway, the issue looks like false positive to me.

closed 2019-08-27 by jukkar