zephyrproject-rtos/zephyr #18424: [Coverity CID :203489]Memory - corruptions in /tests/application_development/gen_inc_file/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/application_development/gen_inc_file/src/main.c
 Category: Memory - corruptions
 Function: test_gen_gz_inc_file_mtime_arg
 Component: Tests
 CID: 203489
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-08-18:
@marc-hb from 3061c92

aescolar, 2019-08-18:
Related to #18425

marc-hb, 2019-08-19:
After a few seconds https://scan9.coverity.com/reports.htm#v32951/p12996 redirects to https://scan9.coverity.com/reports.htm#v38298 which shows a spinning wheel for a few more seconds and then leaves a blank page. The top right corner shows my email address even though I didn't explicitly logged in, at least not in the last few months. The bottom left corner shows "0 projects match". This could be an awkward and falsely "secure" way to tell me I don't have access? I just requested access at https://scan.coverity.com/memberships/new?project_id=zephyr, hope it helps.

PR #18532 (2019-08-21): tests/ /gen_inc_file: convert 4,5,6,7 enumeration to an interval
It's shorter and may get rid of false Coverity positives 203489 and
203498 and fix #18424 and duplicate #18425.
commit a3da6389 (2019-08-21): tests/ /gen_inc_file: convert 4,5,6,7 enumeration to an interval

It's shorter and may get rid of false Coverity positives 203489 and
203498 and fix #18424 and duplicate #18425.
closed 2019-08-21 by galak