zephyrproject-rtos/zephyr #18435: [Coverity CID :203481]API usage errors in /tests/crypto/tinycrypt/src/test_ecc_utils.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/crypto/tinycrypt/src/test_ecc_utils.c
 Category: API usage errors
 Function: string2scalar
 Component: Tests
 CID: 203481
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-08-18:
from e2ad85b696e #784
@pfl you changed this statement in 21358baa72c, would be so kind as to correct it?

```
 71void string2scalar(unsigned int *scalar, unsigned int num_word32, char *str)
 72{
 73
 74        unsigned int num_bytes = num_word32 * 4U;
 75        uint8_t tmp[num_bytes];
 76        size_t hexlen = strlen(str);
 77
 78        int padding;
 79
 80        padding = num_bytes * 2U - strlen(str);
 81        if (0 > padding) {
 82                printf("Error: 2 * num_bytes(%d) < strlen(hex) (%zu)\n",
   	
CID 203481 (#1 of 1): Invalid type in argument to printf format specifier (PRINTF_ARGS)
invalid_type: Argument strlen(str) to format specifier %zu was expected to have type size_t(unsigned long) but has type unsigned int. [Note: The source code implementation of the function has been overridden by a builtin model.]
 83                       num_bytes * 2U, strlen(str));
 84                k_panic();
 85        }
 86
 87        (void)memset(tmp, 0, padding / 2);
 88
 89        if (hex2bin(str, hexlen, tmp + padding / 2, num_bytes) == 0) {
 90                k_panic();
 91        }
 92        uECC_vli_bytesToNative(scalar, tmp, num_bytes);
 93
 94}
```

pfl, 2019-08-19:
> @pfl you changed this statement in [21358ba](https://github.com/zephyrproject-rtos/zephyr/commit/21358baa72cea9c23be57444eb91444774842f97), would be so kind as to correct it?

I changed the next line to add an 'U' to the unsigned constant multiplying the other argument, the size_t strlen() was never touched. The only header file I can find is:
`libc/minimal/include/string.h:extern size_t strlen(const char *s);`
which according to my understanding is most certainly printed with modifier '%zu'. Where is this unsigned long coming from, do you use another compiler as gcc, or is strlen() re/mis-defined for this case somehow? What is the architecture where this appears? I get no errors with '-DBOARD=native_posix'.

aescolar, 2019-08-19:
@pfl 
>  do you use [..] What is the architecture where this appears?

To the best of my understanding, none, as this is a Coverity static analysis. This has nothing to do with native_posix.

From the error it is also not clear to me what the problem is.

nashif, 2020-03-11:
marked as fixed in coverity.

closed 2020-03-11 by nashif