zephyrproject-rtos/zephyr #18440: [Coverity CID :203439]Memory - corruptions in /tests/kernel/mem_protect/protection/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/kernel/mem_protect/protection/src/main.c
 Category: Memory - corruptions
 Function: write_text
 Component: Tests
 CID: 203439
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

andyross, 2019-08-21:
Same as the other three I just looked at.  False positive.  Coverity doesn't like memory reads or writes to symbols in the .text section (it thinks their sizes are zero), but that is exactly the behavior we are testing.

andyross, 2019-08-23:
Assign to @aasthagr as I don't know how to manage recording this as a false positive in the tool