zephyrproject-rtos/zephyr #18441: [Coverity CID :203460]Memory - corruptions in /tests/kernel/mem_protect/protection/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/kernel/mem_protect/protection/src/main.c
 Category: Memory - corruptions
 Function: execute_from_buffer
 Component: Tests
 CID: 203460
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

andyross, 2019-08-21:
False positive.   Similar to the other two I've seen where the tool is faced with pointer to code: it assumes the symbol size is zero.  This is undefined behavior per C, but it's required by the test so that we can exercise memory reads from .text and validate memory protection.

andyross, 2019-08-23:
Assign to @aasthagr as I don't know how to manage recording this as a false positive in the tool

ioannisg, 2019-09-23:
@aasthagr could you please mark this as fixed/false-alarm?

aasthagr, 2019-11-12:
Marked as false positive in Coverity

closed 2019-11-12 by aasthagr