zephyrproject-rtos/zephyr #18442: [Coverity CID :203465]Memory - corruptions in /tests/kernel/mem_protect/protection/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/kernel/mem_protect/protection/src/main.c
 Category: Memory - corruptions
 Function: write_text
 Component: Tests
 CID: 203465
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

andyross, 2019-08-21:
Just like #18444, this is a false positive.  The test is deliberately attempting to write to the .text section and verify that the failure gets detected by the memory protection implementation.  Static analysis is correct that this is invalid, but that's the point.

andyross, 2019-08-23:
Assign to @aasthagr as I don't know how to manage recording this as a false positive in the tool

ceolin, 2019-08-30:
I've done it. Just closing it.

closed 2019-08-30 by ceolin