zephyrproject-rtos/zephyr #18444: [Coverity CID :203516]Memory - corruptions in /tests/kernel/mem_protect/userspace/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/kernel/mem_protect/userspace/src/main.c
 Category: Memory - corruptions
 Function: write_kerntext
 Component: Tests
 CID: 203516
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203516:  Memory - corruptions  (OVERRUN)
/tests/kernel/mem_protect/userspace/src/main.c: 255 in write_kerntext()
249     static void write_kerntext(void)
250     {
251     	/* Try to write to kernel text. */
252     	expect_fault = true;
253     	expected_reason = K_ERR_CPU_EXCEPTION;
254     	BARRIER();
>>>     CID 203516:  Memory - corruptions  (OVERRUN)
>>>     Overrunning buffer pointed to by "&z_is_thread_essential" of 0 
>>> bytes by passing it to a function which accesses it at byte offset 3 
>>> using argument "4U". [Note: The source code implementation of the 
>>> function has been overridden by a builtin model.]
255     	memset(&z_is_thread_essential, 0, 4);
256     	zassert_unreachable("Write to kernel text did not fault");
257     }
258     
259     static int kernel_data;
260 
```

andyross, 2019-08-21:
This is a false positive.  The code in question is **deliberately trying** to do an invalid write (i.e. by writing four bytes to the beginning of the call address of the z_is_thread_essential() function).  The tool is right to complain, but the behavior is exactly as intended.

It might be possible to change the expression here in a way that the tool wouldn't detect, but I don't think that's worth the effort.

andyross, 2019-08-23:
Assign to @aasthagr as I don't know how to manage recording this as a false positive in the tool

ceolin, 2019-08-30:
done

closed 2019-08-30 by ceolin