zephyrproject-rtos/zephyr #18447: [Coverity CID :203400]Integer handling issues in /tests/lib/fdtable/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/lib/fdtable/src/main.c
 Category: Integer handling issues
 Function: test_z_finalize_fd
 Component: Tests
 CID: 203400
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
f1afb4c24d0
```
 62void test_z_finalize_fd(void)
 63{
 64        const struct fd_op_vtable *vtable;
 65
   	1. negative_return_fn: Function z_reserve_fd() returns a negative number. [show details]
   	2. assign: Assigning: fd = z_reserve_fd().
 66        int fd = z_reserve_fd();
 67        int *obj = z_get_fd_obj_and_vtable(fd, &vtable);
 68
 69        const struct fd_op_vtable *original_vtable = vtable;
 70        int *original_obj = obj;
 71
   	
CID 203400 (#1 of 1): Improper use of negative value (NEGATIVE_RETURNS)
3. negative_returns: fd is passed to a parameter that cannot be negative. [show details]
 72        z_finalize_fd(fd, obj, vtable); /* function being tested */
 73
 74        obj = z_get_fd_obj_and_vtable(fd, &vtable);
 75
 76        zassert_equal_ptr(obj, original_obj, "obj is different after finalizing");
 77        zassert_equal_ptr(vtable, original_vtable, "vtable is different after finalizing");
 78
 79        z_free_fd(fd);
 80}
```

nashif, 2019-12-10:
was fixed in afe5b2598d0a64c3943dfc507445d334a3c95f39

closed 2019-12-10 by nashif