zephyrproject-rtos/zephyr #18450: [Coverity CID :203505]Integer handling issues in /tests/lib/fdtable/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/lib/fdtable/src/main.c
 Category: Integer handling issues
 Function: test_z_reserve_fd
 Component: Tests
 CID: 203505
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203505:  Integer handling issues  (NEGATIVE_RETURNS)
/tests/lib/fdtable/src/main.c: 18 in test_z_reserve_fd()
12     void test_z_reserve_fd(void)
13     {
14     	int fd = z_reserve_fd(); /* function being tested */
15     
16     	zassert_equal(fd >= 0, true, "fd < 0");
17     
>>>     CID 203505:  Integer handling issues  (NEGATIVE_RETURNS)
>>>     "fd" is passed to a parameter that cannot be negative.
18     	z_free_fd(fd);
19     }
20     
21     void test_z_get_fd_obj_and_vtable(void)
22     {
23     	const struct fd_op_vtable *vtable;
```

aescolar, 2019-11-11:
f1afb4c24d0

PR #21086 (2019-12-02): tests: fdtable: check for negative fd
Check for negative file descriptor.

Fixes #18448
Fixes #18450
Fixes #18449
Fixes #18451
commit afe5b259 (2019-12-02): tests: fdtable: check for negative fd

Check for negative file descriptor.

Fixes #18448
Fixes #18450
Fixes #18449
Fixes #18451
closed 2019-12-02 by nashif