zephyrproject-rtos/zephyr #18451: [Coverity CID :203528]Integer handling issues in /tests/lib/fdtable/src/main.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/lib/fdtable/src/main.c
 Category: Integer handling issues
 Function: test_z_alloc_fd
 Component: Tests
 CID: 203528
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aescolar, 2019-11-11:
f1afb4c24d0
```
 82void test_z_alloc_fd(void)
 83{
 84        const struct fd_op_vtable *vtable = NULL;
 85        int *obj = NULL;
 86
   	1. negative_return_fn: Function z_alloc_fd(obj, vtable) returns a negative number. [show details]
   	2. assign: Assigning: fd = z_alloc_fd(obj, vtable).
 87        int fd = z_alloc_fd(obj, vtable); /* function being tested */
 88
 89        obj = z_get_fd_obj_and_vtable(fd, &vtable);
 90
   	3. Condition (void *)obj == NULL, taking true branch.
   	4. Condition (void *)obj == NULL, taking true branch.
   	5. Condition "obj is different after allocating", taking true branch.
   	6. Condition "obj is different after allocating", taking true branch.
 91        zassert_equal_ptr(obj, NULL, "obj is different after allocating");
   	7. Condition (void *)vtable == NULL, taking true branch.
   	8. Condition (void *)vtable == NULL, taking true branch.
   	9. Condition "vtable is different after allocating", taking true branch.
   	10. Condition "vtable is different after allocating", taking true branch.
 92        zassert_equal_ptr(vtable, NULL, "vtable is different after allocating");
 93
   	
CID 203528 (#1 of 1): Improper use of negative value (NEGATIVE_RETURNS)
11. negative_returns: fd is passed to a parameter that cannot be negative. [show details]
 94        z_free_fd(fd);
 95}
```

aescolar, 2019-11-11:
@nashif Who in Intel takes this one?

PR #21086 (2019-12-02): tests: fdtable: check for negative fd
Check for negative file descriptor.

Fixes #18448
Fixes #18450
Fixes #18449
Fixes #18451
commit afe5b259 (2019-12-02): tests: fdtable: check for negative fd

Check for negative file descriptor.

Fixes #18448
Fixes #18450
Fixes #18449
Fixes #18451
closed 2019-12-02 by nashif