zephyrproject-rtos/zephyr #18459: [Coverity CID :203509]Null pointer dereferences in /tests/subsys/usb/desc_sections/src/desc_sections.c

aasthagr, 2019-08-17:
Static code scan issues seen in File: /tests/subsys/usb/desc_sections/src/desc_sections.c
 Category: Null pointer dereferences
 Function: check_endpoint_allocation
 Component: Tests
 CID: 203509
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

aasthagr, 2019-08-17:
```
*** CID 203509:  Null pointer dereferences  (FORWARD_NULL)
/tests/subsys/usb/desc_sections/src/desc_sections.c: 191 in check_endpoint_allocation()
185     			struct usb_ep_descriptor *ep_descr =
186     				(struct usb_ep_descriptor *)head;
187     
188     			/* Check that we get iface desc before */
189     			zassert_not_null(cfg_data, "Check available cfg data");
190     
>>>     CID 203509:  Null pointer dereferences  (FORWARD_NULL)
>>>     Passing null pointer "cfg_data" to "find_cfg_data_ep", which dereferences it.
191     			zassert_true(find_cfg_data_ep(ep_descr, cfg_data,
192     						      ep_count),
193     				     "Check endpoint config in cfg_data");
194     			ep_count++;
195     		}
196     
```

jfischer-phytec-iot, 2019-08-21:
Seems to be false positive, there is `zassert_not_null(cfg_data, "Check available cfg data");` to check cfg_data pointer.

aescolar, 2019-08-22:
@jfischer-phytec-iot are you able to mark it as a false positive in Coverity?

carlescufi, 2019-08-28:
Marked as false positive in Coverity

closed 2019-08-28 by carlescufi