zephyrproject-rtos/zephyr #18473: RFC: List of Zepyhr Consultancy companies

rettichschnidi, 2019-08-18:
## Introduction

The projects website should have a compilation of companies offering Zephyr-related services. 

### Problem description

Companies typically have money and product ideas and developers. But given the relatively young age of the Zephyr project, they typically lack experience with Zephyr.
Getting started with Zephyr while avoiding the pitfalls of creating a (mass) product is hard, but can be simplified greatly with the help of more experienced Zephyr developers/partners.

### Proposed change

The website of Zephyr should contain an new section with a compilation of companies and freelancers doing consultancy/engineering work for Zephyr.

## Detailed RFC

The new section on the Zephyr website should contain a compilation of companies which offer consultancy and engineering work related to Zephyr.

Proposed information to be displayed for every company:
* Offered services
* Past projects (hardware, product, scale, ...)
* Relevant contributions (e.g. maintaining a specific area of Zephyr, commits, hardware support)
* Spoken languages, time zone
* Website and/or contact information
* Some free text field

To allow this compilation to thrive and thus helping the adoption of Zephyr, it should be free of charge to be listed.

### Dependencies
* People willing to add companies (to get the compilation started)
* Companies willing to add and maintain their own entries (in the long run)

### Concerns and Unresolved Questions
* There should be a mechanism to prevent outdated information from staying on the website

## Alternatives
* Every company interested in getting started with Zephyr having a hard(er) time finding a fitting partner.

rettichschnidi, 2019-08-18:
Potential candidates I could find so far:
* [Prowess Consulting](https://www.prowesscorp.com/)
  * Source: [Developing a Neonatal Baby Monitor](https://www.prowesscorp.com/developing-a-neonatal-baby-monitor/)

mike-scott, 2019-08-19:
Could also include the company's primary expertise.  For instance, Foundries.io focuses on secure device management such as LwM2M to enable OTA via 802.15.4/OpenThread, BLE 6LoWPAN and LTE-M/NB-IoT.

carlescufi, 2019-08-19:
I wonder if this should be on the website or directly in our documentation. Maybe this is something worth discussing at the next TSC?
@MaureenHelm @rettichschnidi @nashif

carlescufi, 2019-08-19:
> Potential candidates I could find so far:
> 
> * [Prowess Consulting](https://www.prowesscorp.com/)
>   
>   * Source: [Developing a Neonatal Baby Monitor](https://www.prowesscorp.com/developing-a-neonatal-baby-monitor/)
> * [Foundries.io](http://foundries.io/)
>   
>   * Source: Spoke with them on ELCE 2018
> * [3mdeb](https://3mdeb.com/)
>   
>   * Source: Talked to them on Embedded World 2019; https://blog.3mdeb.com/tags/zephyr/
> * [Wind River](https://www.windriver.com/)
>   
>   * Source: [Wind River Rocket—Better Together with Zephyr Project](https://www.windriver.com/products/operating-systems/rocket/)
> * [TrustAZUR](http://www.trustazur.com/)
>   
>   * Source: [Services](http://www.trustazur.com/services/) (click on "Custom Software Development")
> * [Antmicro](https://antmicro.com/)
>   
>   * Source: [Soft­ware & BSPs](https://antmicro.com/services/software-and-bsps/)

I would probably add there:

- [CodeCoup](https://www.codecoup.pl/)
- [Bay Libre](https://baylibre.com/)
- @pabigot (not sure he has a website)

jfischer-phytec-iot, 2019-08-19:
> I would probably add there:
>
>     * [CodeCoup](https://www.codecoup.pl/)
>
>     * [Bay Libre](https://baylibre.com/)
>
>     * @pabigot (not sure he has a website)

Well then I would add [PHYTEC Messtechnik GmbH](https://www.phytec.de).

pfalcon, 2019-08-19:
Guys, you can get a bit of advertising by posting articles to Zephyr blog: https://www.zephyrproject.org/news/blog/ ;-).

pfalcon, 2019-08-19:
We can also publish a stats report which companies contributed which % of commits to Zephyr release X.Y, as popular for Linux, e.g. https://lwn.net/Articles/726950/

rettichschnidi, 2019-08-19:
> I wonder if this should be on the website or directly in our documentation. Maybe this is something worth discussing at the next board meeting?

I'd be willing to join and (if helpful) maintain such a compilation.

pabigot, 2019-08-19:
> * @pabigot (not sure he has a website)

I do, though it's rarely updated and I'm not actively seeking new clients at this time.  I'd like to be on the list nonetheless.

* [Peter Bigot Consulting, LLC](https://www.pabigot.com/)

dbkinder, 2019-08-20:
I'm notifying the Zephyr marketing committee chair about this so they can incorporate this into a Zephyr website update in the works.

mimeriza, 2019-08-21:
Thanks for this excellent suggestion and for dbkinder's heads up. The Zephyr marketing committee will definitely take a look at adding something like this to the Zephyr website (which is being redesigned even as we speak.)

tautologyclub, 2019-08-27:
The company I work for, [Endian Technologies](https://endian.se), are quite excited about Zephyr and we will probably do a fair bit of upstreaming in the future. We've hosted a Zephyr hackathon and we're currently using Zephyr in multiple projects.

carlescufi, 2019-10-11:
> We're always interested in telling the story of people using Zephyr. Is there a contact I should engage with?

You can join for a conference such as Embedded  Linux Conference (US or Europe) and submit a talk proposal there.

mbolivar, 2019-10-11:
>I wonder if this should be on the website or directly in our documentation

Just my two cents, but documentation is forever -- old versions are preserved. This seems to be more of a momentary piece of information (who's consulting *now*?).

dbkinder, 2019-10-11:
… I’ve brought this request to the attention of the Marketing folks who manage the zephyrproject.org site, showing as an example, what was done for the Yocto Project:
   https://www.yoctoproject.org/community/consultants/

-- david

bfrog, 2019-10-17:
I'd also like to be listed

Tom Burdick
[Electromaticus](https://electromatic.us)

parthitce, 2020-07-02:
Please include [Linumiz](https://www.linumiz.com/) - parthiban@linumiz.com

mturquette, 2020-07-03:
Thanks @carlescufi for mentioning BayLibre! We're happy to be on the list. only nitpick is that our corporate name is a single word: BayLibre

I know I know, it's CaMeLcAsE but that's marketing for you

Is there any update on when such a list might go live on the Zephyr Project website?

rettichschnidi, 2020-07-07:
How about having a way for a consultant to specify if having commit permissions and/or willing to do PR reviews for money? Would allow companies to get a better chance to get their commits merged soonish.

carlescufi, 2020-10-05:
@rettichschnidi How about we move the information you have in [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/18473#issuecomment-522350121) over to the body of this issue? You could even turn it into a table, feel free to use [this tool](https://www.tablesgenerator.com/markdown_tables) if it makes it easier for you. Then we can add more info to the table, such as the one you propose. Once we have a nice table we can again raise the issue of moving this to the website at the TSC level, but in the meantime we'll have something that our users can use more easily.

carlescufi, 2020-10-05:
@rettichschnidi also please add @gmarull to the future table: https://teslabs.com/

MaureenHelm, 2020-10-05:
We now have a services page: https://zephyrproject.org/member-offerings/

Please contact @kestewart about getting listed.

rettichschnidi, 2020-10-05:
@carlescufi I like the idea of having a proper table. However, what about creating a page in the Wiki instead of this issue (where only me and some maintainers?) can edit it?

@MaureenHelm Will it be possible for non-members (e.g. independent contractors, small companies, etc.) to be listed on this page? If so, I'm in favor of closing this issue.

MaureenHelm, 2020-10-06:
> Any chance it would be possible for non-members (e.g. independent contractors, small companies, etc.) to be listed on this page? If so, I'm in favor of closing this issue.

@kestewart could you please comment?

kestewart, 2020-10-19:
@MaureenHelm @carlescufi @rettichschnidi - we're discussing adding this sort of page for this to the website next year.  There's a fair amount of process to figure out so it doesn't end up with stale and misleading data.

carlescufi, 2020-10-19:
> @MaureenHelm @carlescufi @rettichschnidi - we're discussing adding this sort of page for this to the website next year. There's a fair amount of process to figure out so it doesn't end up with stale and misleading data.

In that case I agree with @rettichschnidi that a page on the Wiki would perhaps be the best interim solution. Any objections @kestewart or @MaureenHelm ?

kestewart, 2020-10-19:
@carlescufi - concern is who is going to confirm that they agree to be listed,  as well as what happens if someone raises an issue,  so if we're documenting it on the WIKI,  we'd want the process documented, and someone to be owner to curate the page, etc.

carlescufi, 2020-10-20:
> @carlescufi - concern is who is going to confirm that they agree to be listed, as well as what happens if someone raises an issue. If we're documenting it on the WIKI, we'd want the process documented there too, and someone to be owner to curate the page, etc.

@kestewart agreed. I could put together a proposal to document the process in the very same wiki page that lists the companies. I will work towards that.

pietrushnic, 2021-03-12:
Hi all,
we (3mdeb) are getting more and more requests for Zephyr RTOS consultancy especially in light of Google announced they will use Zephyr for [EC in Chromebooks](https://cfp.osfc.io/osfc2020/talk/SXYUQD/). and I wonder if there is any plan to push effort discussed in this issue. If we cannot support that somehow please let us know.

cfriedt, 2021-11-03:
Friedt Professional Engineering Services, Inc
(website is under construction, but I could list telephone / email)

mbolivar-nordic, 2022-01-11:
Can this be closed now that we have https://www.zephyrproject.org/ecosystem-vendor-offerings/?

rettichschnidi, 2022-01-12:
Seems like non-members are also able to get listed there. Thanks!

closed 2022-01-12 by rettichschnidi