zephyrproject-rtos/zephyr #18476: Custom module with west

Laczen, 2019-08-19:
Hi,

I am trying to create a custom module (e.g. mymodule). I have created a folder mymodule that contains mymodule.c, mymodule.h, CMakeLists.txt and a zephyr subfolder containing module.yml.

CMakeLists.txt:
```
# SPDX-License-Identifier: Apache-2.0
if(CONFIG_MODULE)
add_library(mymodule INTERFACE)

target_include_directories(mymodule INTERFACE ${CMAKE_CURRENT_SOURCE_DIR})

zephyr_library()
zephyr_library_sources(
    mymodule.c
    )
zephyr_library_link_libraries(mymodule)
target_link_libraries(mymodule INTERFACE zephyr_interface)
endif()
```

module.yml:
```
build:
  cmake: .
```

To use the module my app CMakeLists.txt is:
```
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.13.1)
set(ZEPHYR_MODULES abspath/mymodule [...])
include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)

project(test_mymodule)

FILE(GLOB app_sources src/*.c)

target_sources(app PRIVATE ${app_sources})
```

When I try to build using west I can see that the module is included, but the include file mymodule.h is not added to the include directories, and compilation fails at `#include <mymodule.h>`.

What change is needed in the module's CMakeLists.txt to add the include file to the includepath?

mbolivar, 2019-08-19:
> set(ZEPHYR_MODULES abspath/mymodule [...])

> When I try to build using west

Not sure if related to this issue, but there is support for a ZEPHYR_EXTRA_MODULES variable which can be used to add a fixed set custom modules to the ones which are found via west and which will populate ZEPHYR_MODULES if it is not already set.

If your ZEPHYR_MODULES list in the `[...]` isn't complete, you might run into other issues, and you also risk version skew as zephyr changes. I'd recommend using the EXTRA version.

Laczen, 2019-08-19:
@mbolivar, thanks for your reply, I have changed ZEPHYR_MODULES into ZEPHYR_EXTRA_MODULES, but the result remains the same. It does seem better for the existing modules as they are all added now.

carlescufi, 2019-08-19:
Two things to take into account:

1.  If you are using west then there's no need for the `set(ZEPHYR_MODULES ...` in your app's CMakeListst.txt, west will populate that automatically
2. Try using `zephyr_include_directories()` in your module's CMakeLists.txt

Laczen, 2019-08-19:
@carlescufi, thanks for the reply. The `zephyr_include_directories()` solves my problem.

I was trying to find a general approach for a custom local module, and keeping the include file in the module instead of in the zephyr tree. Maybe there is a better way to add this custom local module to west and use the standard `target_include_directories()`.
What is the way to add a custom module to west in the case you would like to keep everything in your local module directory (include files, Kconfig, ...) ?

carlescufi, 2019-08-19:
> @carlescufi, thanks for the reply. The `zephyr_include_directories()` solves my problem.
>
> I was trying to find a general approach for a custom local module, and keeping the include file in the [...]

I am really not sure. All the main CMakeLists.txt does is `add_subdirectory()` to each module folder, so anything you use outside of a module should work here too. Maybe @SebastianBoe can add some more info.

SebastianBoe, 2019-08-19:
> When I try to build using west I can see that the module is included, but the include file mymodule.h is not added to the include directories, and compilation fails at #include <mymodule.h>.

@Laczen : Perhaps instead of

`target_include_directories(mymodule INTERFACE ${CMAKE_CURRENT_SOURCE_DIR})`

do

`zephyr_include_directories(.)`

Laczen, 2019-08-19:
@SebastianBoe, it works with `zephyr_include_directories()`, thanks.

mbolivar, 2019-08-19:
> 1.  If you are using west then there's no need for the `set(ZEPHYR_MODULES ...` in your app's CMakeListst.txt, west will populate that automatically

@carlescufi I think "using west" is ambiguous here btw and I want to
clear that up in case others are reading this issue.

"Using west build" != "put the custom module in west.yml"

IMO:

- If the custom module is not in west.yml, ZEPHYR_EXTRA_MODULES should be used.
- If it is in west.yml, no need to touch ZEPHYR_EXTRA_MODULES.

Laczen, 2019-08-19:
@SebastianBoe, It seems I was to quick with stating that it works. In my case my header file sources some zephyr headers and this seems to fail. Is it possible to create a ZEPHYR_EXTRA_MODULES that uses zephyr headers ?
@mbolivar, the term "using west" is indeed ambiguous, I was referring to the command used for building.

SebastianBoe, 2019-08-20:
> Is it possible to create a ZEPHYR_EXTRA_MODULES that uses zephyr headers ?

Yes. AFAICT the code in the description should work, could you publish the code in some branch so I can reproduce?

Laczen, 2019-08-20:
@SebastianBoe, you are correct, something else was wrong with my code that made me believe the include files were missing. I am closing the issue, thanks for the help.

closed 2019-08-20 by Laczen