zephyrproject-rtos/zephyr #18483: Bluetooth: length variable inconsistency in keys.c

jhedberg, 2019-08-19:
After commit 94451b22fcc there's an inconsistency with length variable usage:
https://github.com/zephyrproject-rtos/zephyr/blob/c635f4cb6d56492620114ade116c9b85b20bb9e6/subsys/bluetooth/host/keys.c#L339-L341

Is the if-statement wrong in using `len_rd` or is the log message wrong in using `len`?

aescolar, 2019-08-20:
Unfortunately @fnde-ot is out for the next 2 weeks. But he had left this comment about this: https://github.com/zephyrproject-rtos/zephyr/commit/94451b22fcc6a068de2a2982b202772c6bbefae8#r34438851

aescolar, 2019-08-20:
@jhedberg ^^

aescolar, 2019-08-20:
@jhedberg I bounce this issue back at you as we cannot expect @fnde-ot to react any time soon.

aescolar, 2020-02-14:
Fixed as a sideeffect of #19407

closed 2020-02-14 by aescolar