zephyrproject-rtos/zephyr #18488: Bluetooth: Mesh: Friend queue message seqnum order

trond-snekvik, 2019-08-19:
**Describe the bug**
The friend queue messages are encrypted when they're put into the queue.  This causes replay protection issues on the LPN if we get a friend control message we have to respond to: 

Friend subscription list add/remove messages should be responded to with a subscription list confirm message immediately, essentially skipping the queue. This functionality is correctly implemented in [`enqueue_sub_cfm`](https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/bluetooth/mesh/friend.c:434), but the enqueued messages from the friend node itself allocate their sequence numbers as they're added to the queue (in `bt_mesh_friend_enqueue_tx`). Later, when the LPN does a poll for enqueued messages, it'll reject the enqueued messages in the replay protection, as it already received a higher seqnum in the subscription list confirm message, that skipped the queue.

**To Reproduce**
1. Establish a friendship between an LPN and a friend node.
2. Enqueue one or more app level messages from the friend to the LPN.
3. Add a group subscription to the LPN, triggering a Friend subscription list add message
4. After the confirm is received from the friend, do an LPN poll. All the messages from the friend to the LPN should be rejected by the replay protection.

**Impact**
Unexpected communication breakdown between the Friend and the LPN. 

**Required changes**
To get around this, I believe the only viable solution is to store friend queue messages unencrypted, and only allocate sequence numbers and encrypt once we know the message will be sent without anything cutting the line (ie when we get a poll and that message is first in the queue).

jhedberg, 2019-08-19:
We might need to store some more meta-data if we store unencrypted, e.g. which IV Index to use (must be the same as when the Friend received the PDU). Or you think the IVI bit in the header is enough for this?

jhedberg, 2019-08-19:
Just to be clear: I'm all for storing unencrypted, in fact this is hinted as the "right thing" by looking at section "3.11 Message processing flow" in the mesh spec.

trond-snekvik, 2019-08-19:
That's true, but we essentially do this already, just encrypted and obfuscated. I think IVI is enough, the message can't stay in the queue for more than 96H, according to the note in 3.6.6.3.2.

carlescufi, 2019-08-20:
@jhedberg @trond-snekvik plans to submit a PR for this in time for 2.0?

jhedberg, 2019-08-20:
> @jhedberg @trond-snekvik plans to submit a PR for this in time for 2.0?

@carlescufi I can't commit to it right now at least. I would also combine this change with a more light-weight TID-based tracking of messages in the queue, rather than using the SeqAuth + address combination.

trond-snekvik, 2019-08-20:
I agree, this is too big of a change to do in just a few days.

trond-snekvik, 2019-08-26:
I'll take a stab at this :)

PR #18745 (2019-09-16): Bluetooth: Mesh: Encrypt friend packets on send
Stores friend queue packets unencrypted, removing any out-of-order
issues caused by seqnum allocation. Also moves as much of the metadata
storage as possible into the packet, allowing us to free up some bytes
of net_buf user data for friend packets.

Fixes #18488
commit 532241d5 (2019-09-16): Bluetooth: Mesh: Encrypt friend packets on send

Stores friend queue packets unencrypted, removing any out-of-order
issues caused by seqnum allocation. Also moves as much of the metadata
storage as possible into the packet, allowing us to free up some bytes
of net_buf user data for friend packets.

Fixes #18488
closed 2019-09-16 by jhedberg