zephyrproject-rtos/zephyr #18508: tests/net/trickle failed on frdm_k64f board

chen-png, 2019-08-20:
**Describe the bug**
Running test suite test_trickle on frdm_k64f board, it failed with error message "test_trickle_2_wait: (cb_2_called is false)".

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=frdm_k64f  tests/net/trickle
2. west flash

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.10.2
 - Commit ID: 59bbf3a521

ioannisg, 2019-08-23:
@chen-png are you looking into this issue?

ioannisg, 2019-08-23:
Test fails also on sam_e70_xplained, so this is not platform-specific.

nashif, 2019-08-23:
duplicate of #18598

closed 2019-08-23 by nashif