zephyrproject-rtos/zephyr #18522: BLE: Mesh: When transport send seg_msg to LPN

menglingao1997, 2019-08-20:
https://github.com/zephyrproject-rtos/zephyr/blob/07460176ed9edb731e5ffee775c7c09259ee33d8/subsys/bluetooth/mesh/transport.c#L314-L465
```C
        if (IS_ENABLED(CONFIG_BT_MESH_FRIEND) && 
 	    !bt_mesh_friend_queue_has_space(tx->sub->net_idx, net_tx->src, 
 					    tx->dst, &tx->seq_auth, 
 					    tx->seg_n + 1) && 
 	    BT_MESH_ADDR_IS_UNICAST(tx->dst)) { 
 		BT_ERR("Not enough space in Friend Queue for %u segments", 
 		       tx->seg_n + 1); 
 		seg_tx_reset(tx); 
 		return -ENOBUFS; 
 	} 
```

According to the code above, if I send a SEG message (greater than CONFIG_BT_MESH_FRIEND_QUEUE_SIZE) to a UNICAST Address (not an LPN address),will be cause an abnormality.

I think should add some functions to determine if it is sent to LPN, such as `bt_mesh_friend_match`

jhedberg, 2019-08-20:
@menglingao1997 I don't see a bug here. Could be because of bad naming of the APIs, but if you look at the implementation of `bt_mesh_friend_queue_has_space()` you'll see that it'll return `true` if no LPN matches the given address, i.e. this branch will not be taken in that case.

jhedberg, 2019-08-20:
@menglingao1997 here's the code I'm referring to:
https://github.com/zephyrproject-rtos/zephyr/blob/07460176ed9edb731e5ffee775c7c09259ee33d8/subsys/bluetooth/mesh/friend.c#L1280-L1285

I'll close this, but please reopen if you think something is still buggy or unclear.

jhedberg, 2019-08-20:
Just wanted to add that one of the reasons I didn't go for a separate "match" call is that I'm trying to avoid iterating the LPN list too many times. We're already doing it twice for TX and three times for RX, which I'm not very happy about.

menglingao1997, 2019-08-20:
https://github.com/zephyrproject-rtos/zephyr/blob/07460176ed9edb731e5ffee775c7c09259ee33d8/subsys/bluetooth/mesh/friend.c#L1256-L1264
if seg num bigger than CONFIG_BT_MESH_FRIEND_QUEUE_SIZE, above function return false, this is my concerned.

jhedberg, 2019-08-20:
> if seg_count bigger than CONFIG_BT_MESH_FRIEND_QUEUE_SIZE, above function return false, this is my concerned.

@menglingao1997 that's a real issue, thanks for clarifying. I'm thinking we might be able solve this through Kconfig so that the Friend Queue size must be at least as big as the maximum number of segments we can receive.

jhedberg, 2019-08-20:
FWIW, e.g. with the default settings this is a non-issue, since that's 3 RX segments max and 16 for the friend queue size.

I also realised it's not so trivial with Kconfig since the RX segment count is derived indirectly from BT_MESH_RX_SDU_MAX

jhedberg, 2019-08-20:
@menglingao1997 I think I'll just fix it like this:
```
--- a/subsys/bluetooth/mesh/friend.c
+++ b/subsys/bluetooth/mesh/friend.c
@@ -1226,6 +1226,10 @@ static bool friend_queue_has_space(struct bt_mesh_friend *frnd, u16_t addr,
        u32_t total = 0;
        int i;
 
+       if (seg_count > CONFIG_BT_MESH_FRIEND_QUEUE_SIZE) {
+               return false;
+       }
+
        for (i = 0; i < ARRAY_SIZE(frnd->seg); i++) {
                struct bt_mesh_friend_seg *seg = &frnd->seg[i];
 
@@ -1259,10 +1263,6 @@ bool bt_mesh_friend_queue_has_space(u16_t net_idx, u16_t src, u16_t dst,
        bool someone_has_space = false, friend_match = false;
        int i;
 
-       if (seg_count > CONFIG_BT_MESH_FRIEND_QUEUE_SIZE) {
-               return false;
-       }
-
        for (i = 0; i < ARRAY_SIZE(bt_mesh.frnd); i++) {
                struct bt_mesh_friend *frnd = &bt_mesh.frnd[i];
 
```

Can I get a quick ack to proceed?

menglingao1997, 2019-08-20:
I think ok.

PR #18528 (2019-08-20): Bluetooth: Mesh: Fix discarding messages with many segments
The code for checking space in the Friend queue was faulty in the case
that we receive a message with more segments than the configure Friend
Queue size. This is not an issue for the default configuration but
still a possible one. Move the check for exceeding Friend Queue Size
to the per-LPN function, so that bt_mesh_friend_queue_has_space()
iterates all LPNs before delivering its verdict. This allows us to
return success in case no LPN matched (which is how the code was
intended to work).

Fixes #18522
commit 6de5b79f (2019-08-20): Bluetooth: Mesh: Fix discarding messages with many segments

The code for checking space in the Friend queue was faulty in the case
that we receive a message with more segments than the configured Friend
Queue size. This is not an issue for the default configuration but
still a possible one. Move the check for exceeding Friend Queue Size
to the per-LPN function, so that bt_mesh_friend_queue_has_space()
iterates all LPNs before delivering its verdict. This allows us to
return success in case no LPN matched (which is how the code was
intended to work).

Fixes #18522
commit 0d5640ec (2019-08-20): Bluetooth: Mesh: Fix discarding messages with many segments

The code for checking space in the Friend queue was faulty in the case
that we receive a message with more segments than the configure Friend
Queue size. This is not an issue for the default configuration but
still a possible one. Move the check for exceeding Friend Queue Size
to the per-LPN function, so that bt_mesh_friend_queue_has_space()
iterates all LPNs before delivering its verdict. This allows us to
return success in case no LPN matched (which is how the code was
intended to work).

Fixes #18522
closed 2019-08-20 by jhedberg