zephyrproject-rtos/zephyr #18524: No disconnection event during "heavy" indication stream

YuraCobain, 2019-08-20:
After connection is dropped by master in inappropriate manner(for example master is power off) and slave is streaming heavily using indication mechanism no disconnection event is generated and slave become unreachable.

**To Reproduce**
Steps to reproduce the behavior:
1. Slave is Xenon Particle, master Android with nrfConnect app.
2. Setup nonsecure connection.
3. Enable indication on custom characteristic and start stream.
4. Kill nrfConnect app.

Result: 
1. No disconnect event is generated.
2. Slave is not discoverable .

**Expected behavior**
disconnected callback is involved and slave is ready to accept new connection.

**Environment (please complete the following information):**
 - OS:  Linux Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.10.3
 - Zephyr: master @0746017

**Additional context**
Each connection interval slave sent indication of 64 bytes of user payload.

BT configs

CONFIG_BT=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_SMP=y
CONFIG_BT_ATT_PREPARE_COUNT=5
CONFIG_BT_DEVICE_NAME="stargate-sg1-13"
CONFIG_BT_DEVICE_APPEARANCE=833
CONFIG_BT_DEVICE_NAME_DYNAMIC=y
CONFIG_BT_DEVICE_NAME_MAX=65
CONFIG_BT_GATT_BAS=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_BT_PERIPHERAL_PREF_MIN_INT=10
CONFIG_BT_PERIPHERAL_PREF_MAX_INT=14

CONFIG_BT_DEBUG_CONN=y

CONFIG_BT_L2CAP_TX_BUF_COUNT=8
CONFIG_BT_L2CAP_TX_FRAG_COUNT=4
CONFIG_BT_L2CAP_TX_MTU=230

CONFIG_BT_SETTINGS=y

logs during reproduction:

[no_bt_dis.log](https://github.com/zephyrproject-rtos/zephyr/files/3520622/no_bt_dis.log)

aescolar, 2019-08-20:
@carlescufi could you triage?

YuraCobain, 2019-08-20:
The same behavior even when use proper disconnect request from nrfConnect
Indication is on and packet with any user data is sent every connection interval.

carlescufi, 2019-08-20:
@YuraCobain could you please check the following:

Are you using the latest split controller? Search your `build/zephyr/.config` for `CONFIG_BT_LL_SW_SPLIT`.

 Is this code path being executed:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/hci/hci.c#L3036

maybe add an `LL_ASSERT(0)` and check if it's triggered?

If it's not, then please check if those are being processed:
https://github.com/zephyrproject-rtos/zephyr/blob/01d11d50d3f98e1da31d71b947458a579335de86/subsys/bluetooth/controller/ll_sw/ctrl.c#L10071
or
https://github.com/zephyrproject-rtos/zephyr/blob/07460176ed9edb731e5ffee775c7c09259ee33d8/subsys/bluetooth/controller/ll_sw/ull_conn.c#L3209

YuraCobain, 2019-08-21:
@ioannisg regarding your questions:

1. By default those configs are on in .config
BLE Controller support

CONFIG_BT_LL_SW_SPLIT=y
CONFIG_BT_LL_SW_LEGACY is not set
CONFIG_BT_LLL_VENDOR_NORDIC=y

2. In case of active indication stream any of mentioned code paths are executed.
    Stream using notification works properly.

aescolar, 2019-08-21:
@carlescufi ^^

carlescufi, 2019-08-21:
> @ioannisg regarding your questions:
> 
> 1. By default those configs are on in .config
>    BLE Controller support
> 
> CONFIG_BT_LL_SW_SPLIT=y
> CONFIG_BT_LL_SW_LEGACY is not set
> CONFIG_BT_LLL_VENDOR_NORDIC=y
> 
> 1. In case of active indication stream any of mentioned code paths are executed.
>    Stream using notification works properly.

@YuraCobain I see. And can you verify that when disconnection works properly the code paths I mentioned are executed? Just to make sure that you are adding the right ASSERT statements in the right places.

YuraCobain, 2019-08-21:
Sure, when I drop connection during notification stream ASSERTS are triggered in hci.c and ll_conn.c.

carlescufi, 2019-08-21:
> Sure, when I drop connection during notification stream ASSERTS are triggered in hci.c and ll_conn.c.

Thanks. Could you please switch to the legacy controller and see if the symptoms are the same?
```
CONFIG_BT_LL_SW_SPLIT=n
CONFIG_BT_LL_SW_LEGACY=y
```

YuraCobain, 2019-08-21:
With legacy controller the symptoms are the same.

YuraCobain, 2019-08-23:
@carlescufi  and @ioannisg  here is the simple scenario to reproduce the bug.
For "samples/bluetooth/peripheral_ht/src/main.c" replace main function by
`
void main(void)
{
	int err;

	err = bt_enable(bt_ready);
	if (err) {
		printk("Bluetooth init failed (err %d)\n", err);
		return;
	}

	bt_conn_cb_register(&conn_callbacks);
	bt_conn_auth_cb_register(&auth_cb_display);
    

	/* Implement indicate. At the moment there is no suitable way
	 * of starting delayed work so we do it here
	 */
	while (1) {
		k_sleep(K_MSEC(50));

		/* Temperature measurements simulation */
		hts_indicate();

		/* Battery level simulation */
		//bas_notify();
	}
}
`
1. Connect to the "Zephyr Health Thermometer" via nrfConnect app on Android.
2. Enable indication.
3. Disconnect gracefully. 
Result:
No disconnect event, no advertisement from slave.

joerchan, 2019-08-23:
@YuraCobain Thanks, Will try this out.

carlescufi, 2019-08-23:
@YuraCobain I have a feeling this might be an issue on the phone side. Maybe when there's indications pending the phone doesn't disconnect the link. Do you have another, non-Zephyr peripheral that you can use to test so that we verify if this is the case?

joerchan, 2019-08-23:
Maybe it is something wrong with my phone, but from what I'm seeing I cannot disconnect when indication is on-going (interestingly everytime I press the disconnect button it does a conn param update).

Killing the nRF connect app also does not result in link disconnecting.

@YuraCobain What phone do you use?
Do you have the chance to confirm with a bluetooth sniffer that link is in fact disconnected and no disconnected callback was seen?

YuraCobain, 2019-08-23:
@carlescufi are right it's Android issue.
On Huawei p20 smart the connection is keep active when indication is on-going.
Even disable Bluetooth does not work and only reboot of phone actually close the connection.
On Samsung Galaxy S8 connection is closed after Bluetooth is disabled.
On IPhone the issue is not reproducible.

joerchan, 2019-08-23:
Ok. then we have the same phone, and are seeing the same thing.

Closing as this is an Android issue.

closed 2019-08-23 by joerchan