zephyrproject-rtos/zephyr #18530: Convert GPIO drivers to new GPIO API

galak, 2019-08-20:
This is just a tracking issue on status and who will look at converting GPIO drivers to support new GPIO API:

- [ ] atmel_sam3
- [ ] sam0
- [ ] cc2650
- [ ] esp32
- [ ] pcal9535a
- [ ] sifive
- [ ] stellaris
- [ ] ht16k33
- [ ] altera_nios2
- [ ] qmsi
- [ ] qmsi_ss
- [ ] dw
- [ ] intel_apl
- [ ] mchp_xec
- [ ] sch
- [ ] cmsdk_ahb
- [ ] mmio32
- [ ] stm32
- [ ] cc13xx_cc26xx
- [ ] cc32xx

@pabigot 
- [ ] sx1509b

@mnkp 
- [ ] sam
- [ ] gecko
- [ ] nrfx

@MaureenHelm:
- [ ] rv32m1
- [ ] imx
- [ ] mcux
- [ ] mcux_igpio
- [ ] mcux_lpc

galak, 2019-08-20:
@nategraff-sifive I assume you are good with looking to convert the sifive gpio driver to the new GPIO api.

galak, 2019-08-20:
@henrikbrixandersen I assume you are good with looking to convert the ht16k33 to the new GPIO api.

henrikbrixandersen, 2019-08-21:
> @henrikbrixandersen I assume you are good with looking to convert the ht16k33 to the new GPIO api?

@galak Sure.

pabigot, 2019-09-04:
### Proposed Roadmap to GPIO Integration for #15611

This is a slightly edited description of stages originally proposed on slack.  The goal is to do development and review work on a topic branch, but to merge pieces to master as soon as they are sufficiently ready that their presence will not break any tests or applications.  There are seven stages:

* **(1)** Add the new API and test code as in #16648 plus pieces of #18689, and enough additional drivers to be convinced the approach is sound.  These merge to the topic branch when ready. Existing code uses old API and old implementation and should work correctly.

  It is not a goal of this stage to convert all drivers, only the ones that have an active responsible codeowner who commits to doing the conversion in a timely manner (e.g. within the first 2-3 weeks of 2.1).  We must expose the work to the entire community quickly so we get feedback on issues before significant effort is wasted.  There will be time to make corrections.
* **(2)** Add a revertible commit that makes deprecation messages configurable.  Merge the initial capability on topic branch to master with this commit set to disable warnings.  All tests should pass.

  The topic branch is rebased on the resulting master, and adds a commit to re-enable deprecation warnings so areas that need attention are easily identifiable.
* **(3)** Re-implement the old API using the new basis functions in a single commit on the topic branch.  All tests should pass for applications where all GPIO peripherals have been converted, if deprecation warnings are disabled.  Tests for systems that have not implemented the new API will fail.
* **(4)** Implement the new API on remaining platforms, each in a single commit for the peripheral driver and all target-specific devicetrees. These are reviewed and tested on the topic branch.  When one is deemed stable it gets merged to master to join the others.

  Simultaneously any driver that uses GPIO may be updated to the new API; examples include sensors and networking hardware.  These commits stay on the topic branch until anything that might use them have converted to the new implementation, at which point they can merge to master.
* **(5)** When all SOC/external GPIO peripherals have been converted the old API implementation functions are removed.  This is validated on the topic branch, then merged to master.
* **(6)** Any remaining GPIO uses can be cleaned up by commits targeting master.  This can begin as of stage **(4)** and proceed independently as long as they do not introduce failures on unconverted platforms.  Issues may be opened identifying particular sensors/drivers/applications where conversion must be done and validated by the codeowner.
* **(7)** Once all such uses are converted the ability to control deprecation of the old API is removed on master and the integration for GPIO changes is complete.

Material is merged to master at/during stages **(2)** and **(4)** through **(6)**.  During those stages the topic branch works for converted systems/applications, but may not build for others.  Master should pass sanitycheck at all times and be bisectable except possibly between commits within a merge, just as is true for existing PRs.

This approach requires active management: cherry-picking pieces from the topic branch into PRs that target master, and frequent rebasing of the topic branch.  This is not technically difficult to do, and if this approach is approved I can take on that task.  Success requires responsiveness from stakeholders who approve the topic-branch merge to quickly approve the master merge.

It also requires responsiveness from stakeholders outside the API meeting participants: vendors to convert GPIO peripheral drivers in **(4)** and other codeowners to update and validate code that uses the GPIO API in **(6)**.

I would like to see us reach stage **(4)** no later than the end of September.  I believe this is achievable.

#18689 has been a prototype of this: it contains material that is similar to master at stage **(4)**.    sanitycheck will pass in that situation (as long as CI doesn't break, as it did in my last push).  I've also got commits that verify subsequent when restricted to converted devices; they are currently not in that PR.

galak, 2019-09-04:
Would be good to get a summary of changes that need to be made to the driver and dts files.

pabigot, 2019-09-04:
IMO the best approach is to use the five existing conversions as examples, but:

Essentially for the driver it's "implement the new functions in the API table, and make the tests pass".

For the devicetree files it's minimally replace the deprecated macros and update the flags to be correct for the corresponding device (generally leave at 0 for active high, or change to `GPIO_ACTIVE_LOW` if appropriate) so the updated LED and button examples will work and driver code can use `gpio_pin_set` without worrying about logical/physical conversions.

mnkp, 2019-09-09:
I've added "Steps required to convert a GPIO driver to the new API" section directly to the issue description so it's easy to spot. Please let me know if there are parts which are not clear / should be extended.

pabigot, 2019-09-11:
### Proposed GPIO policies for interrupt use/safety, atomicity, etc. related to #18970

_Copied from slack, edited here based on feedback_

* All functions must be invokable from an ISR.  Functions that cannot accomplish their roles in that context may return `-EWOULDBLOCK`.
* It must be possible to disable a GPIO pin interrupt from a GPIO interrupt callback.
* All pin configuration functions (GPIO, interrupt, handler) must be atomic.
* All functions that change port outputs must be atomic.  (For example, gpio_port_toggle() must lock in any system that doesn't have a hardware toggle register.)

pabigot, 2019-09-11:
The potential issue with the existing API is that the way to disable interrupts is `gpio_pin_configure_interrupt()`.  The requirements in https://github.com/zephyrproject-rtos/zephyr/issues/18530#issuecomment-530472454 are easily met for SOC GPIO drivers.  For an external GPIO chip like SX1509B disabling a specific pin interrupt would require an I2C transaction, which cannot be done from an ISR.

Currently SX1509B doesn't support interrupts.  If it does, it could still satisfy the above requirements because I carefully worded them to allow an implementation that flagged the interrupt, but didn't invoke the callback within the ISR.  This could still cause functional problems (the situation where I usually want to disable interrupts in the callback is where I'm using a level trigger, and the callback can't perform the actions necessary to clear or change the level).

The questions are (a) do we care about interrupts from external GPIO devices, and (b) if we do is there any API change that would resolve this, e.g. defining `gpio_pin_interrupt_{enable,disable}()` distinct from the interrupt configuration?

I think the answer should be (a) yes, and (b) no, which leads to (c) we need to document that callbacks for some GPIO drivers may be invoked from a helper thread rather than within the ISR itself.

galak, 2019-10-29:
@jenmwms any update on the intel-apl or pcal9535a drivers?

jenmwms, 2019-10-29:
@Kumar Gala<mailto:kumar.gala@linaro.org> I am working on the intel-apl. It could help to have a pair programming sync with someone familiar with apl. I’ll reach out to those around today.  I’m making progress though!
@Leung, Daniel<mailto:daniel.leung@intel.com> is looking at the pcal9535a drivers.

pabigot, 2020-01-28:
All drivers are converted on the topic branch.

closed 2020-01-28 by pabigot