zephyrproject-rtos/zephyr #18546: Hard Fault when connecting to BLE device

ChristianHirsch, 2019-08-21:
**Describe the bug**
nRF51822 (CFAC) crashes when connecting to it several times. Android (7.1.1) app connects to device to write (initialize timer) and read data (measurements). After a couple of iterations (connecting, reading the data, disconnecting) Zephyr crashes. It is not always the same number of iterations.

**To Reproduce**
TBA

**Impact**
It is needed to reinitialize the hardware, measurements in RAM are lost - showstopper

**Screenshots or console output**
`
FATAL: ***** HARD FAULT *****
FATAL: r0/a1:  0x00000003  r1/a2:  0x20004630  r2/a3:  0x00000000
FATAL: r3/a4:  0x00000002 r12/ip:  0x000186e3 r14/lr:  0x00002a3f
FATAL:  xpsr:  0x01000011
FATAL: Faulting instruction address (r15/pc): 0x000108be
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x200015ec (unknown)
[00:04:49.397,247] <err> bt_ctlr_llsw_nordic_lll_conn: assert: 'e' failed
FATAL: Halting system
`

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.10.3, west v0.6.0
 - Zephyr v2.0.0-rc1 (4524035bc8d5e8eb892bccc1ce9547b09e9c7f52)

**Additional context**
TBA

carlescufi, 2019-08-21:
@ChristianHirsch thanks for reporting this. We are currently investigating this crash

ChristianHirsch, 2019-08-21:
@carlescufi thanks for investigating! If you need more information, please tell me, I'll try to help.

Unfortunately, I don't see a pattern when this happens. Sometimes it succeeds with 9+ iterations (connecting, discovering services, receiving data, disconnecting), but sometimes it manages just 3 iterations. It seems to be random..

What I can tell: it happens right after creating a connection. The `connected` callback function in Zephyr is called (I see the debug output), but in Android the connection is not established. It aborts with **Status 133** `GATT ERROR`. Normally, I start discovering the services, right after creating a connection, but I also tried to add a delay between connection establishment and discovering services without success.

PR #18569 (2019-08-22): Bluetooth: controller: split: Fix Tx FIFO, queue, and pool corruption
Fix the Tx FIFO, queue and pool corruption due to missing
release of link object to Tx link pool and hence missing
reset of the per connection initially allocated Tx link
free pointer.

The bug caused Tx PDUs and associated memory to be lost
leading to missing L2CAP segment transmissions. With lost
control PDU buffers, ULL would stall processing Done events
also leading to controller asserts.

Fixes #18546.
commit bdcec841 (2019-08-22): Bluetooth: controller: split: Fix Tx FIFO, queue, and pool corruption

Fix the Tx FIFO, queue and pool corruption due to missing
release of link object to Tx link pool and hence missing
reset of the per connection initially allocated Tx link
free pointer.

The bug caused Tx PDUs and associated memory to be lost
leading to missing L2CAP segment transmissions. With lost
control PDU buffers, ULL would stall processing Done events
also leading to controller asserts.

Fixes #18546.
closed 2019-08-22 by carlescufi