zephyrproject-rtos/zephyr #18554: Tracking Issue for C++ Support as of release 2.0

pabigot, 2019-08-21:
Issues that impact C++ support as of Zephyr release 2.0

## Not yet worked or need review:
- [ ] #1205 involves need for explicit cast from `void *` to `type *` in C++ builds
- [ ] #4028 requests new/delete, vectors, RTTI, exceptions, static global dtor.  Some of this is done.
- [ ] #5695 mostly done, but blocked by #18551 
- [ ] #16239 trivial? need to excluded `-Wold-style-definition` from C++ builds
- [ ] #18551 address-of-temporary idiom illegal in C++

## PR pending
- [ ] #15603 cannot include_next stdhdr.h ; pr #18552 
- [ ] no issue, relates to heap alloc with real libc++ ; pr #18464

## Resolved

pabigot, 2019-08-21:
@galak @vanwinkeljan FYI.

aescolar, 2019-08-22:
@pabigot Thanks for collecting this list. If you will be working on this towards 2.1 it would be nice if you could comment it in one of the coming TSC meetings so others are made aware of it.

brooksprumo, 2020-01-14:
Hi all, I found a C++ issue with mcuboot.h, which was missing `extern "C"` guards.  I've created a PR for it here:

https://github.com/zephyrproject-rtos/zephyr/pull/21916

I'm not sure if this is the right place to track this PR, so please let me know if I should comment elsewhere.

cfriedt, 2021-04-14:
A while ago I started looking at taking the conventional route to support std::thread, std::mutex, and std::condition_variable (basically wrapping all of it around pthread using gthr.h). Would involve minor modifications to the toolchain. Wrapping pthread requires pthread_create to accept a NULL attr, which requires dynamically allocated thread stacks, ..., dynamic kobject, heap allocator modifications. I'm hopefully at the bottom of that rabbit hole now 😅

A lot of it will be necessary without even using pthread, but wrapping pthread would be a useful shortcut even though it might add to the footprint. Definitely interested in helping out with this effort.

nashif, 2021-05-28:
2.1 is long gone.

closed 2021-05-28 by nashif