zephyrproject-rtos/zephyr #18568: Support for Particle Photon

scott-eddy, 2019-08-22:
Is there any planned support for the [particle photon](https://store.particle.io/collections/wifi/products/photon)?  I know that #11176 and subsequent #12195 added support for the xenon, argon, and boron but I am unable to see anything about the photon.  I know that the board is fundamentally different as it is based around a STM32F205 instead of a nRF52840, but support for it would be nice.  

I also know that photon support has been added to [NuttX](https://bitbucket.org/nuttx/nuttx/pull-requests/260/configs-add-particle-photon-board-support/diff), another excellent RTOS.  

Is there any planned or ongoing work on this I am not aware of?  If not I can look into adding support myself.

galak, 2019-08-22:
Adding @pabigot as he added the other particle boards so worth checking with him on the photon.

I'm guessing the base photon support wouldn't be too difficult as we already have support for an STM32F207 SoC & board, so figuring the STM32F205 isn't going to be too different.

pabigot, 2019-08-22:
I did the original mesh feathers because I had them from work on a Nordic-based C++ sensor platform.  I have no plans to add photon, so it's all yours.

carlescufi, 2020-03-24:
Please send a Pull Request.

closed 2020-03-24 by carlescufi