zephyrproject-rtos/zephyr #18570: Dynamic interrupt does not work with multi-level interrupts

dcpleung, 2019-08-22:
There is a potential for writing into memory past `_sw_isr_table[]` due to the interrupt encoding with multi-level interrupts and dynamic interrupts. This is because `z_isr_install()` in `arch/common/sw_isr_common.c` simply takes the input `irq` to calculate a linear index to `_sw_isr_table[]`. However, in reality, the elements after 1st level interrupts are sparsely populated and no longer linear. For example:

* Each level has 32 IRQs.
* There is one 2nd level interrupts out of #6 of 1st level.
* So the interrupt encoding of the 7th IRQ on 2nd level would be '0x0806'.
* `z_isr_install()` sets `_sw_isr_table[2054]` with the ISR.
* However, due to there is only one 2nd level interrupt, the index should be 39 (32 + 7).
* Also note that the size of `_sw_isr_table` is only 64 (32 + 32).

aescolar, 2019-08-22:
CC @andyross @andrewboie

dcpleung, 2020-09-15:
Fixed in commit 0fb43821644fa2912390195fd5295bc3199ebb19

closed 2020-09-15 by dcpleung