zephyrproject-rtos/zephyr #18574: Some platforms: “reel_board”, “frdm_k64f” and “sam_e70_xplained”  are be built failure

xuhao8210, 2019-08-22:
**Describe the bug**
Some platforms: “reel_board”, “frdm_k64f” and “sam_e70_xplained”  are be built failure, but I can build successfully on commit ID:” 2d9d898cc6a309d87e831aa913da3de43e026c75”. It seems that the failure may be caused by commit ID: cf3af6af5fa3e656d6caf0b3429fde52e895fa79 

**To Reproduce**
Steps to reproduce the behavior:
1.	Change folder to zephyr 
2.	Source zephyr-env.sh.
3.	Run the west build command  , e.g. command : ” west build -b reel_board samples/hello_world/”
4.	They are failure when build platforms: “reel_board”, “frdm_k64f” and “sam_e70_xplained”.

**Expected behavior**
All platforms can be built successfully via west build command. 

**Impact**
Some platforms: “reel_board”, “frdm_k64f” and “sam_e70_xplained”  are be built failure on commit ID: d157527ec1304d5d289bdb079974b6e48aed54d7
**Screenshots or console output**
[ztest@zserver1 zephyr]$ west build -b reel_board samples/hello_world/
-- west build: build configuration:
       source directory: /home/ztest/work/zephyrproject/zephyr/samples/hello_world
       build directory: /home/ztest/work/zephyrproject/zephyr/build
       BOARD: reel_board (origin: command line)
-- west build: generating a build system
Zephyr version: 2.0.0
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4")
-- Selected BOARD reel_board
-- Found west: /home/ztest/.local/bin/west (found suitable version "0.6.0", minimum required is "0.6.0")
-- Loading /home/ztest/work/zephyrproject/zephyr/boards/arm/reel_board/reel_board.dts as base
-- Overlaying /home/ztest/work/zephyrproject/zephyr/dts/common/common.dts
Device tree configuration written to /home/ztest/work/zephyrproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/ztest/work/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ztest/work/zephyrproject/zephyr/boards/arm/reel_board/reel_board_defconfig'
Merged configuration '/home/ztest/work/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/ztest/work/zephyrproject/zephyr/build/zephyr/.config'
-- Cache files will be written to: /home/ztest/.cache/zephyr
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ztest/zephyr-sdk-0.10.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
Including module: esp-idf in path: /home/ztest/work/zephyrproject/modules/hal/esp-idf/zephyr
Including module: fatfs in path: /home/ztest/work/zephyrproject/modules/fs/fatfs
Including module: qmsi in path: /home/ztest/work/zephyrproject/modules/hal/qmsi
Including module: cypress in path: /home/ztest/work/zephyrproject/modules/hal/cypress
Including module: silabs in path: /home/ztest/work/zephyrproject/modules/hal/silabs
Including module: st in path: /home/ztest/work/zephyrproject/modules/hal/st
Including module: stm32 in path: /home/ztest/work/zephyrproject/modules/hal/stm32
Including module: libmetal in path: /home/ztest/work/zephyrproject/modules/hal/libmetal
Including module: lvgl in path: /home/ztest/work/zephyrproject/modules/lib/gui/lvgl
Including module: mbedtls in path: /home/ztest/work/zephyrproject/modules/crypto/mbedtls
Including module: mcumgr in path: /home/ztest/work/zephyrproject/modules/lib/mcumgr
Including module: nffs in path: /home/ztest/work/zephyrproject/modules/fs/nffs
Including module: open-amp in path: /home/ztest/work/zephyrproject/modules/lib/open-amp
Including module: openthread in path: /home/ztest/work/zephyrproject/modules/lib/openthread
Including module: segger in path: /home/ztest/work/zephyrproject/modules/debug/segger
Including module: tinycbor in path: /home/ztest/work/zephyrproject/modules/lib/tinycbor
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ztest/work/zephyrproject/zephyr/build
-- west build: building application
[6/114] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
ccache /home/ztest/zephyr-sdk-0.10.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=v2.0.0-rc1-161-gd157527ec130 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../kernel/include -I../arch/arm/include -I../include -I../include/drivers -Izephyr/include/generated -I../soc/arm/nordic_nrf/nrf52 -I../soc/arm/nordic_nrf/include -I../ext/hal/cmsis/Include -isystem ../lib/libc/minimal/include -isystem /home/ztest/zephyr-sdk-0.10.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /home/ztest/zephyr-sdk-0.10.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Os -imacros/home/ztest/work/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacros/home/ztest/work/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/ztest/work/zephyrproject/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ztest/work/zephyrproject/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj   -c /home/ztest/work/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c
In file included from ../include/arch/arm/cortex_m/cmsis.h:17,
                 from ../include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h:10,
                 from ../include/arch/arm/cortex_m/mpu/arm_mpu.h:13,
                 from ../include/arch/arm/arch.h:246,
                 from ../include/arch/cpu.h:17,
                 from ../include/kernel_includes.h:34,
                 from ../include/kernel.h:17,
                 from ../kernel/include/kernel_structs.h:10,
                 from /home/ztest/work/zephyrproject/zephyr/arch/arm/core/offsets/offsets.c:26:
../soc/arm/nordic_nrf/nrf52/soc.h:17:10: fatal error: nrf.h: No such file or directory
 #include <nrf.h>
          ^~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /usr/bin/cmake --build /home/ztest/work/zephyrproject/zephyr/build

**Environment (please complete the following information):**

OS: Fedora release 28 
Commit ID :  d157527ec1304d5d289bdb079974b6e48aed54d7

jfischer-phytec-iot, 2019-08-22:
`west update` ?

xuhao8210, 2019-08-22:
@jfischer-phytec-iot  Thanks for your reminding, after I input command:"west update", it can be built successfully. few files were been moved to module folder. (e.g. modules/hal/nordic/nrfx/mdk/nrf.h). So close it.

closed 2019-08-22 by xuhao8210