zephyrproject-rtos/zephyr #18579: Add generic Type-C "port manager" support

raveslave, 2019-08-22:
**Add support for USB Type-C controllers with PD**

This is useful when designing products using USB-C connectors needing additional support like higher current, role-switching "host/device", power source/sink or simply supporting vendor packages over PD (using the CC pins).

The driver should implement support for
- DRP (DFP/UFP), i.e. the type-c terms for "OTG", with low amount of programmability.
- PD 2.0 / 3.0 to support source or sink, incl providing or negotiating power-requirements (voltage/current) with PD-compliant power-supplies or computers using Type-C.
- Alternate mode(s)

The driver should basically serve as a state machine that orchestrates connection-events, data and power roles as well as runtime changes like role-swap or power-updates.

The connection between the TCPM (running in Zephyr) and the TCPC (vendor IC) is defined as the USB Type-C Port Controller Interface, TCPCI. 
see: https://www.usb.org/document-library/usb-type-cr-port-controller-interface-specification

**Event and control needs**

The low level driver should be responsible for
- Reporting VBUS status and activating VBUS
- Setting CC lines and providing CC line status
- Setting line polarity
- Activating and deactivating VCONN
- Setting current limits
- Activating and deactivating PD message transfers
- Sending and receiving PD messages
- wip...

---   
[image]

keith-zephyr, 2021-07-27:
Google is in the planning process of upstreaming the [USB-C stack](https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/platform/ec/common/usbc/) used on Chromebooks into Zephyr.

The is a bit of work needed to reorganize the code to conform to Zephyr's structure, so if anyone has specific features like would like prioritized, please add a comment to this issue.

dzarda, 2021-08-02:
@keith-zephyr That's a nice bit of news. Last time I checked there was no Sink-only support in the Chrome OS codebase. However that's a major case for Zephyr devices. Would that be available?

keith-zephyr, 2021-08-02:
Yes, we plan to provide source-only, sink-only, and dual-role port operation.  We also support try source and try sink operation for dual-role ports.

keith-zephyr, 2021-09-07:
This [github issue](https://github.com/zephyrproject-rtos/zephyr/issues/38371) has been created that will cover the work of moving Google's USB-C stack to Zephyr.   We welcome comments and suggestions.

gmsdesai87, 2026-09-23:
Thanks, this clarifies the TCPM/TCPC boundary. I have one follow-up question related to a USB-C phone-to-phone power-sharing use case.
Suppose an application wants to implement a user-controlled "Send Battery" function where:
Device A selects a target amount of energy to transfer to Device B;
Device B accepts the request;
the system establishes the appropriate source/sink and PD configuration;
the transfer is monitored; and
the transfer terminates when the agreed energy target has been reached.
In the Zephyr/Linux architecture, which layer would normally be responsible for:
receiving the application's requested energy-transfer target;
translating that target into the appropriate PD/power policy;
measuring or determining how much energy has actually been delivered; and
terminating the transfer when the target is reached?
In particular, is there an existing standard/API mechanism for an application to request a quantified energy-transfer target (for example Wh or an equivalent battery-energy target), or does the application normally only request/affect power roles and electrical operating parameters?
I'm trying to understand whether a user-level application can realistically sit above TCPM and control a quantified peer-to-peer energy-transfer transaction, or whether that requires OEM/platform-specific integration.

keith-zephyr, 2026-09-23:
> In particular, is there an existing standard/API mechanism for an application to request a quantified energy-transfer target (for example Wh or an equivalent battery-energy target), or does the [...]

I'm not aware of anything in the PD specification that tracks or communicates information about cumulative energy.  So this kind of policy needs to be tracked and managed by the application.

If the application is monitoring the energy transfer (or more likely just monitoring the current battery charge), the sink device can stop accepting current by disabling the `SinkVbus` in the TCPC.  The PD contract remains in place so VBUS remains at the negotiated voltage, but the sink device just doesn't accept current.  The Zephyr TCPC API `tcpc_set_snk_ctrl()` can be used to enable and disable sinking.

For systems that have multiple USB-C ports (such as a laptop), it's common that you disable VBUS sinking as a default.  Only after evaluating the PD contracts on all ports (if connected), the application then selects the "best" port to charge from and only enables VBUS sinking on the specific port.  Enabling VBUS sinking on multiple ports is likely to cause damage to the device.