zephyrproject-rtos/zephyr #18583: hci_usb: NRF52840 connecting addtional peripheral fails

LeBlue, 2019-08-22:
**Describe the bug**
When trying to connect an addtional peripheral, while another is already connected and exchanging data, one the following may happen (for me roughtly 75%).
1. Most of the time the new connection fails to be established in addition to failing to read on the first connected peripheral.
2. The first connected peripheral fails to read characteristic values. The second connection attempt succeeds but still fails to read data (service discovery is never finished).

This error does **not** occur, if both peripherals are connected **before** exchanging data. After both connections are established, data transfer works as expected.

I tested the same peripherals with a BCM dongle, where this problem does not occur.

**To Reproduce**
Steps to reproduce the behavior on the nRF52840_pca10056 (all 3 boards):
1. usb_hci example on one board with CONFIG_BT_MAX_CONN increased to 10
2. use the bluetooth peripheral example on two other boards 
3. Connect one peripheral and read characterisic values continuously
4. Connect the other peripheral and read characteristics values.

**Expected behavior**
(An) additional peripheral(s) can be connected while another exchanges data. The data exchange may still be delayed while the connection procedure is performed.

**Impact**
This is a showstopper as addtional peripherals must be connected, while other are already exchanging data. 

**btmon output**
Log of the failure where the first connected peripheral fails to read data after the second is connected:
[btmon-log-Frist_peripheral_fails_to_read_pca10056.log](https://github.com/zephyrproject-rtos/zephyr/files/3530302/btmon-log-Frist_peripheral_fails_to_read_pca10056.log)
Log of the succes of the same steps with a BCM BT USB dongle
[btmon-log_bcm_dongle_two_peripheral_success.log](https://github.com/zephyrproject-rtos/zephyr/files/3530303/btmon-log_bcm_dongle_two_peripheral_success.log)

Log of the success of the same steps with pca10056 and BT_DEBUG enabled (see Additional context)
[btmon-log_dual_peripheral_success_pca10056_BT_DEBUG.log](https://github.com/zephyrproject-rtos/zephyr/files/3530399/btmon-log_dual_peripheral_success_pca10056_BT_DEBUG.log)

**Environment:**
 - OS: Linux with Bluetooth subsystem version 2.22 and bluez5.50
 - Toolchain (macos/gcc-arm-none-eabi 7.3.1)
 - zephyr v1.14.0

**Additional context**
1. hci_usb was also tested with zephyr version v1.14.1-rc2, with the same errors
2. hci_usb was also tested with zephyr version v1.14.1-rc2, with BT_DEBUG, BT_DEBUG_HCI, BT_DEBUG_ATT to get some logging. **The error was not reproducable any more**, therefore no such logs attached here.

carlescufi, 2019-08-22:
Thanks for the report, we'd need a couple more details to understand:

1. You are always using the `hci_usb` sample and using BlueZ right?
2. Have you tried the current master instead of the 1.14 release?
3. I can't see a connection failing to be established in your logs, which one contains it?
4. I am not sure where to find the "failure to read" either, can you point us to it?

LeBlue, 2019-08-22:
1. Yes, always hci_usb sample and bluez
2. Just working on trying master@aed767a98
3. + 4. I am actually not quite sure, as this might be a bluez issue. Hoped you could provide some intel. I am sure of not issuing the disconnect commands via the dbus API (command #178 in the perhipheral_fails_to_read log). The connection failure notice is issued from bluez dbus API.

I will have to do some more testing, but since it worked always with another usb dongle, i assumend first that it is not a bluez issue.

carlescufi, 2019-08-23:
@LeBlue thanks for the information.
@jhedberg and @Vudentz would you mind checking the attached logs to see if you find anything suspicious on the controller side? I didn't see anything on first inspection

carlescufi, 2019-08-27:
@LeBlue could you please try the master branch and see if this is still an issue? there are quite a few fixes there and a new controller, so it would be really good to know if this fails with master too.

LeBlue, 2019-08-28:
@carlescufi I will test master, but i probably will not have time to do it in the next few days.

Vudentz, 2019-09-20:
Is this still an issue with 2.0? I don't see read request after the second connection or is that the issue? Anyway it appears to be on the BlueZ side.

carlescufi, 2019-10-21:
Closing this due to inactivity.  Feel free to reopen if this is still a problem in the latest mater.

ChristianHirsch, 2020-02-28:
@Vudentz I'm experiencing the same issue. I tried different versions (1.14.1, 2.1.0, 2.2.0-rc2) but they all behave the same.. How can I check if this is a Zephyr or a BlueZ problem? My problem is the following: I'm connecting to two (identical) devices and request some data with:
```bash
gatttool -t random -b AA:BB:CC:DD:EE:FF --char-read --handle=0x0006 ; gatttool -t random -b FF:EE:DD:CC:BB:AA --char-read --handle=0x0006
```
Then, the following happens:
```bash
Characteristic value/descriptor: 00 
connect error: Software caused connection abort (103)
```
After that, device `AA:BB:CC:DD:EE:FF` is disconnected (I see it when scanning) but device `FF:EE:DD:CC:BB:AA` isn't, it is still connected to my PC. From that point on, I cannot connect to any device anymore (at least with `gatttool` I get `connect error: Connection refused (111)`). After restarting the bluetooth service, device `FF:EE:DD:CC:BB:AA` is disconnected and I can connect again to devices.

This problem, however, does not occur with my PC's internal Bluetooth adapter, only with the Zephyr usb_hci example.

jfischer-phytec-iot, 2020-02-28:
> I'm connecting to two (identical) devices and request some data with:

Can it be reproduced with one of the `samples/bluetooth/peripheral*` samples?

ChristianHirsch, 2020-02-28:
@jfischer-phytec-iot thanks for letting me try. I just figured something out: it occurs with the `samples/bluetooth/peripheral_hr` example, if you make the following modification:
```
#define BT_LE_ADV_CONN_NAME BT_LE_ADV_PARAM(BT_LE_ADV_OPT_CONNECTABLE | \
                                            BT_LE_ADV_OPT_USE_NAME, \
                                            BT_GAP_ADV_SLOW_INT_MIN, \
                                            BT_GAP_ADV_SLOW_INT_MAX)
```
So, I'm using `BT_GAP_ADV_SLOW_INT_MIN` and `BT_GAP_ADV_SLOW_INT_MAX` instead of `BT_GAP_ADV_FAST_INT_MIN_2` and `BT_GAP_ADV_FAST_INT_MAX_2` and the sample is flashed onto a `nrf51_pca10028` board. Then, I read some data (from the battery service, but it does not matter):
```bash
gatttool -t random -b AA:BB:CC:DD:EE:FF --char-read --handle=0x0012 ; gatttool -t random -b FF:EE:DD:CC:BB:AA --char-read --handle=0x0012
```
And I get the following output:
```bash
Characteristic value/descriptor: 62 
connect error: Software caused connection abort (103)
```
The `hci_usb` sample is flashed onto a nRF52840 board. Sometimes both commands are executed successfully, but most of the time the second one is not..

LeBlue, 2020-03-02:
I had not any problems creating multiple (up to 9) connections with the `hci_usb` derived FW since around zephry-v2.1.0-rc1. Since there were some other issues (not related to connection establishment) I tested a lot of different versions/master revisions since then (up to zephry-v.2.2.0-rc2).

With zephyrv.1.14.0/1-rc1 this also worked if other established connections did not exchange data _while_ creating a new one (tested only with up to 4 peripherals). Once all connections and service discovery completed, all peripherals were usable in "parallel".

@ChristianHirsch This probably means it is not the same issue, as you are not reading while setting up a new connection. Try using `btmon` for a little bit more info.

Additionally the connection creation would succeed every time, but some timeout in bluez/linux while performing service discovery would trigger a disconnect command (This also returns the "software caused connection abort" error). The exact reason is just a guess from the logs attached here, since i stopped looking into it, due to lack of time.

Note: all peripherals were running zephyr-2.0.0 and also advertising with the slow settings. I should probably upgrade these to find out if there are issues there.

joerchan, 2020-03-02:
@ChristianHirsch Since this appears to be different problem please create a new issue. And provide an as detailed as possible way to reproduce the problem. 

@LeBlue Thank you for responding. If there is nothing else needed for your original problem could you please close the issue again?

closed 2020-03-02 by LeBlue

ChristianHirsch, 2020-03-05:
Thanks @LeBlue for this information.

@joerchan I opened the issue #23280.