zephyrproject-rtos/zephyr #18584: BT LL crash on LL/CON/ADV/BV-04-C

aescolar, 2019-08-22:
**Describe the bug**
The BLE LL crashes during conformance testing, in LL/CON/ADV/BV-04-C
With a controller only build, split LL.

**Expected behavior**
No crash

**Screenshots or console output**
```
LL/CON/ADV/BV-04-C [Accepting Connection Request after Directed Advertising] test started...
d_01: @00:21:46.802490  [00:21:46.802,490] <dbg> bt_ctlr_llsw_nordic_lll_filter.rpa_refresh_start: 
d_02: @00:21:46.843135  [00:21:46.843,109] <dbg> bt_ctlr_llsw_nordic_lll_filter.rpa_refresh_start: 
d_02: @00:21:47.820373  [00:21:47.820,373] <dbg> bt_ctlr_llsw_nordic_lll_filter.rpa_refresh_start: 
d_01: @00:21:48.082489  [00:21:48.082,489] <err> bt_ctlr_llsw_ull_adv: assert: 'adv->link_cc_free' failed
d_01: @00:21:48.082489  @ subsys/bluetooth/controller/ll_sw/ull_adv.c:1134:
d_01: @00:21:48.082489  FATAL: >>> ZEPHYR FATAL ERROR 3: Kernel oops
d_01: @00:21:48.082489  FATAL: Current thread: 0x565df840 (unknown)
d_01: @00:21:48.082489  FATAL: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Commit SHA: aed767a98c00bc11b5455fb60c3a3d8f0bda37c0 (but has been happening for at least 2 weeks)
 - nrf52_bsim with EDTT tests

**Additional context:**
Adding a delay in the middle of the test code avoids the issue (just before re-enabling the scanner)

Labeling as low until it is understood better

aescolar, 2019-08-22:
@cvinayak Does it ring any bell when looking at where it crashes?

aescolar, 2019-08-23:
@carlescufi : Without the asserts it would just dereference a NULL pointer a little bit later (and crash :) )

erbr-ot, 2019-08-23:
I have had a look into this and believe I have identified an issue. There seems to be a race on disable in case of HDC - the ADV_STOP ticker is not stopped in disable() and consequently a start_ticker(....ADV_STOP...) in adv_enable fails, leading to a clean-up of lll->conn data, which in turn leads to the assert when the ADV_STOP ticker expires. 

I have been able to fix this by adding this code to disable():
#if defined(CONFIG_BT_PERIPHERAL)
    ret_cb = TICKER_STATUS_BUSY;
    if (adv->lll.is_hdcd) {
        ret = ticker_stop(TICKER_INSTANCE_ID_CTLR, TICKER_USER_ID_THREAD,
                  TICKER_ID_ADV_STOP,
                  ull_ticker_status_give, (void *)&ret_cb);
        ret = ull_ticker_status_take(ret, &ret_cb);
        if (ret) {
            mark = ull_disable_mark(adv);
            LL_ASSERT(mark == adv);

            return BT_HCI_ERR_CMD_DISALLOWED;
        }
    }
#endif

carlescufi, 2019-08-23:
@erbr-ot great! can you send a PR?

cvinayak, 2019-08-23:
> the ADV_STOP ticker is not stopped in disable() 

Yes, if this is true, ADV_STOP ticker will fire and cause the assert. Please send a PR.

PR #18616 (2019-08-23): bluetooth: controller: Fixing race re. adv disable in HDC
Re. issue "BT LL assert on LL/CON/ADV/BV-04-C #18584"

Fixes #18584
closed 2019-08-23 by carlescufi