zephyrproject-rtos/zephyr #18585: STM32G4 support

rosterloh, 2019-08-22:
**Is your feature request related to a problem? Please describe.**
There is currently no zephyr support for the STM32G4 family

**Describe the solution you'd like**
It would be great if there was support for these. Any sort of timeline for support would be greatly appreciated

erwango, 2019-08-26:
@rosterloh, there is currently no plan to add this series in Zephyr, but for sure it would be welcome!
Help can be provided to any effort to do this porting work.

rosterloh, 2019-08-29:
Ok, I've started work on it. Submitted a PR for the hal and working on the dts

martinjaeger, 2019-10-02:
Here is the PR, if others are looking for it aswell: https://github.com/zephyrproject-rtos/zephyr/pull/18806

erwango, 2019-11-01:
Closing as G4 is supported now. Thanks!

closed 2019-11-01 by erwango