zephyrproject-rtos/zephyr #18587: tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout fails to run on lpcxpresso54114_m4

galak, 2019-08-22:
When running the test via:

`scripts/sanitycheck -p lpcxpresso54114_m4 --device-testing --device-serial  /dev/serial/by-id/usb-SEGGER_J-Link_000723233078-if00 -s tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout`

get the following output:

```
***** Booting Zephyr OS build v2.0.0-rc1-173-gaed767a98c00 *****
Running test suite test_fifo_timeout
===================================================================
starting test - test_timeout_empty_fifo
PASS - test_timeout_empty_fifo
===================================================================
starting test - test_timeout_non_empty_fifo
PASS - test_timeout_non_empty_fifo
===================================================================
starting test - test_timeout_fifo_thread
PASS - test_timeout_fifo_thread
===================================================================
starting test - test_timeout_threads_pend_on_fifo
 thread (q order: 2, t/o: 0, fifo 0x20000008)
 thread (q order: 3, t/o: 10, fifo 0x20000008)
 thread (q order: 0, t/o: 20, fifo 0x20000008)
 thread (q order: 4, t/o: 30, fifo 0x20000008)
 thread (q order: 1, t/o: 40, fifo 0x20000008)
PASS - test_timeout_threads_pend_on_fifo
===================================================================
starting test - test_timeout_threads_pend_on_dual_fifos
 thread (q order: 0, t/o: 0, fifo 0x2000001c)
FAIL - test_multiple_threads_pending@188.  *** thread 3 woke up, expected 1

    Assertion failed at ZEPHYR_BASE/tests/kernel/fifo/fifo_timeout/src/main.c:467: test_timeout_threads_pend_on_dual_fifos: rv not equal to TC_PASS

FAIL - test_timeout_threads_pend_on_dual_fifos
===================================================================
starting test - test_timeout_threads_pend_fail_on_fifo
FAIL - test_multiple_threads_get_data@240.  *** thread 5 woke up, expected 0

    Assertion failed at ZEPHYR_BASE/tests/kernel/fifo/fifo_timeout/src/main.c:488: test_timeout_threads_pend_fail_on_fifo: rv not equal to TC_PASS

    Assertion failed at ZEPHYR_BASE/tests/kernel/fifo/fifo_timeout/src/main.c:131: test_thread_pend_and_timeout: packet == NUFATAL: ***** BUS FAULT *****
FATAL:   Imprecise data bus error
FATAL: r0/a1:  0x20000320  r1/a2:  0x00000000  r2/a3:  0x20003c84
FATAL: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000041d7
FATAL:  xpsr:  0x6100000f
FATAL: Faulting instruction address (r15/pc): 0x00003dc6
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x20000410 (unknown)
FATAL: Halting system
```

galak, 2019-08-22:
Tried on FRDM-K64F and it passes there.

galak, 2019-08-22:
Tested at commit aed767a98c00bc11b5455fb60c3a3d8f0bda37c0

agansari, 2019-08-22:
It's something related to systick, at first suspected that the default CONFIG_SYS_CLOCK_TICKS_PER_SEC=10000 was too fast for this device, but I tested this patch #15487 and there are no more issues.

nashif, 2019-08-29:
duplicate of #18591

closed 2019-08-29 by nashif