zephyrproject-rtos/zephyr #18591: tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout.poll fails to run on FRDM-KW41Z

galak, 2019-08-22:
With commit: aed767a98c00bc11b5455fb60c3a3d8f0bda37c0

Running via:

`./scripts/sanitycheck -p frdm_kw41z --device-testing  --device-serial /dev/serial/by-id/usb-ARM_DAPLink_CMSIS-DAP_0201000028634e45000b5002673a0032a101000097969900-if01 -s tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout.poll`

Get the following output:

```
***** Booting Zephyr OS build v2.0.0-rc1-173-gaed767a98c00 *****
Running test suite test_fifo_timeout
===================================================================
starting test - test_timeout_empty_fifo
PASS - test_timeout_empty_fifo
===================================================================
starting test - test_timeout_non_empty_fifo
PASS - test_timeout_non_empty_fifo
===================================================================
starting test - test_timeout_fifo_thread
PASS - test_timeout_fifo_thread
===================================================================
starting test - test_timeout_threads_pend_on_fifo
 thread (q order: 2, t/o: 0, fifo 0x20000008)
 thread (q order: 3, t/o: 10, fifo 0x20000008)
 thread (q order: 0, t/o: 20, fifo 0x20000008)
 thread (q order: 4, t/o: 30, fifo 0x20000008)
 thread (q order: 1, t/o: 40, fifo 0x20000008)
PASS - test_timeout_threads_pend_on_fifo
===================================================================
starting test - test_timeout_threads_pend_on_dual_fifos
 thread (q order: 0, t/o: 0, fifo 0x2000001c)
FAIL - test_multiple_threads_pending@188.  *** thread 3 woke up, expected 1

    Assertion failed at ZEPHYR_BASE/tests/kernel/fifo/fifo_timeout/src/main.c:467: test_timeout_threads_pend_on_dual_fifos: rv not equal to TC_PASS

FAIL - test_timeout_threads_pend_on_dual_fifos
===================================================================
starting test - test_timeout_threads_pend_fail_on_fifo
FAIL - test_multiple_threads_get_data@240.  *** thread 5 woke up, expected 0

    Assertion failed at ZEPHYR_BASE/tests/kernel/fifo/fifo_timeout/src/main.c:488: test_timeout_threads_pend_fail_on_fifo: rv not equal to TC_PASS

    Assertion failed at ZEPHYR_BASE/tests/kernel/fifo/fifo_timeout/src/main.c:131: test_thread_pend_and_timeout: packFATAL: ***** HARD FAULT *****
FATAL: r0/a1:  0x20000320  r1/a2:  0x00000000  r2/a3:  0x20003c94
FATAL: r3/a4:  0x00000000 r12/ip:  0x20000000 r14/lr:  0x00004697
FATAL:  xpsr:  0x6100000f
FATAL: Faulting instruction address (r15/pc): 0x00004296
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x20000410 (unknown)
FATAL: Halting system

```

agansari, 2019-08-23:
@galak it's similar to what happens on the LPC54xx M0

> It's something related to systick, at first suspected that the default CONFIG_SYS_CLOCK_TICKS_PER_SEC=10000 was too fast for this device, but I tested this patch #15487 and there are no more issues.
> 
> _Originally posted by @agansari in https://github.com/zephyrproject-rtos/zephyr/issues/18587#issuecomment-523934357_

ioannisg, 2019-08-26:
@agansari will you be able to address this for 2.0?
@galak if this is indeed a SysTick driver issue, we need to address this, imho.

ioannisg, 2019-11-20:
This test does not seem to fail any more. I have tested the tests/kernel/fifo/fifo_timeout suite in the following ARM platforms:

| ARM Platform | 
|     :---:      | 
| lpcxpresso54114_m4 |
| frdm_k64f |
| arduino_due |
| nRF5340_dk_nrf5340_cpuapp & w/wo SysTick |
| nRF9160_pca10090 & w/wo SysTIck |
| nRF52_pca10040 & w/wo SysTIck |
| sam_e70_xplained |
| nucleo_g071rb |
| mimxrt1015_evk |

@pabigot could you add some result from other platforms?

pabigot, 2019-11-20:
Here's my conclusions running

    sanitycheck --device-testing --hardware-map all-arm.yml -T tests/kernel/fifo/fifo_timeout

on these platforms:

* atsamd21_xpro (ATML2130021800032454)
* efr32mg_sltb004a (000440128632)
* frdm_k64f (02400201E89F7E40156183F8)
* mimxrt1060_evk (609301751)
* nucleo_l476rg (066FFF535155878281112638)
* nrf52840_pca10056 (000683290670)
* sam_e70_xplained (ATML2637010000000000)

On current master it only fails for:
* atsamd21_xpro: kernel.fifo.timeout.poll

At commit f2b75fd644c87c6cc266a61865b55205cc13b8b9 (which changed how durations were converted, and was a candidate for why it now works on master) the following tests still failed on these platforms:

* atsamd21_xpro: kernel.fifo.timeout kernel.fifo.timeout.poll
* efr32mg_sltb004a: kernel.fifo.timeout kernel.fifo.timeout.poll
* nucleo_l476rg: kernel.fifo.timeout kernel.fifo.timeout.poll
* sam_e70_xplained: kernel.fifo.timeout kernel.fifo.timeout.poll

so that wasn't what fixed it.  Bisecting reveals that most of the failures were eliminated by:
```
commit c724033accdd6df6579652891bb83ea6a9160a19
Author: Anas Nashif <anas.nashif@intel.com>
Date:   Tue Nov 12 09:07:33 2019 -0500

    west: prepend -c to openocd commands
    
    commit 0df4a53107817b080d694ca8a18810d91d498799 changed the behavior of
    how openocd commands are passed to openocd. We used to add -c to each
    command, now the commands are being added without -c causing an error.
    
    This adds "-c" to all commands instead of just passing a list.
    
    Also fixes #20449.
```

This makes absolutely no sense but everything but atsamd21_xpro passes at that commit while they still failed at the previous one.

The final atsamd21_xpro failure is fixed [this commit](https://github.com/zephyrproject-rtos/zephyr/pull/15487/commits/21ab147274002b4cd87dbe57eec34ff5d42605ec) in #15487:

    drivers: systick: fix calculation of absolute cycles count in ISR
    
    The commit fixes the update of the absolute counter of HW cycles
    in the SysTick ISR for TICKLESS mode.

So: Ignoring the inexplicable, we're close now, but we need #15487 to close this.

PR #15487 (2019-11-26): Drivers: ARM: SysTick: investigate errors and apply fixes
Several fixes/improvements in the SysTick timer driver, including
- correct calculation of elapsed cycles in the ISR
- avoiding starving the ISR if timeouts gets re-scheduled all the time
- reworking the way elapsed cycles are calculated

Added some documentation in functions and variables, too.

The fixes are closing several ARM timer-related bugs:

Fixes #18591 which is an instance of #20585
Fixes #20445 which is also an instance of #20585 
Also incorporates the SysTick modification related to #20939

### Mandatory Testing in Hardware [... cut]
closed 2019-11-26 by dleach02