zephyrproject-rtos/zephyr #18592: (nRF51) The RSSI signal does not rise above -44 dBm

Trubochkin, 2019-08-22:
**Describe the bug**
The RSSI signal from the bluetooth beacon does not rise above -44 dBm even when the beacon is at a distance of 1 cm. But at the same time, the value decreases only at a distance of 25 cm or more. I use the NRF51_BLE400 debugging board as the receiver, and the factory beacon on the NRF51822 as the transmitter.

**Expected behavior**
I need the RSSI to change over the entire range of distance. In the android application (nRF_Connect), I see that the RSSI value changes at any distance from the beacon. I can get a value of -20 dBi (at a distance of 1 cm) and below (at a distance). Maybe someone knows what could be the problem?

**Screenshots or console output**
[image]
[image]

**Environment (please complete the following information):**
 - OS: Windows 10
 - Zephyr OS build v2.0.0-rc1

carlescufi, 2019-08-22:
The RSSI is calculated by the phone’s radio and stack, not by Zephyr. I will change this to a question and copy additional people.

cvinayak, 2019-08-23:
> The RSSI is calculated by the phone’s radio and stack, not by Zephyr.

Yes, the value comes from the Radio hardware. Did you try your Zephyr scanner on a nRF51/nRF52  Development Kit?

Trubochkin, 2019-08-23:
> Yes, the value comes from the Radio hardware. Did you try your Zephyr scanner on a nRF51/nRF52 Development Kit?
I tried, but the result is the same.

carlescufi, 2019-08-23:
> > Yes, the value comes from the Radio hardware. Did you try your Zephyr scanner on a nRF51/nRF52 Development Kit?
> 
> Yes

And was it better than the BLE400?

Trubochkin, 2019-08-24:
> And was it better than the BLE400?

No, all the same

Trubochkin, 2019-09-06:
Updated to [v2.0.0rc3](https://github.com/zephyrproject-rtos/zephyr/releases/tag/v2.0.0-rc3) and this solved the problem. Now the value of RSSI rises above -44 dBm 👍

closed 2019-09-06 by Trubochkin