zephyrproject-rtos/zephyr #18593: tests/arch/arm/arm_zero_latency_irqs fails on cc3220sf_launchxl

vanti, 2019-08-22:
**Describe the bug**
Running the test leads to failure to set the test flag.

```
***** Booting Zephyr OS build v2.0.0-rc1-144-g048b0359bcd2 *****
Running test suite zero_latency_irqs
===================================================================
starting test - test_arm_zero_latency_irqs
Available IRQ line: 178

    Assertion failed at ZEPHYR_BASE/tests/arch/arm/arm_zero_latency_irqs/src/arm_zero_latency_irqs.c:72: test_arm_zero_latency_irqs: (post_flag == 1 is false)
Test flag not set by ISR

FAIL - test_arm_zero_latency_irqs
===================================================================
Test suite zero_latency_irqs failed.
===================================================================
PROJECT EXECUTION FAILED
```

**To Reproduce**
Steps to reproduce the behavior:
0. cd tests/arch/arm/arm_zero_latency_irqs
1. mkdir build; cd build
2. cmake .. -DBOARD=cc3220sf_launchxl
3. make flash
4. See error

**Expected behavior**
I would expect the test to pass.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.10.3
 - Commit SHA tested: 1751648db02258d6c56c4ca1e6fea591bf5bc806

ioannisg, 2019-08-23:
I don't have this board to check, but the test does not fail for any other board I have tested it on.

galak, 2019-08-23:
The test fails because it's making an assumption that all IRQs are valid up to CONFIG_NUM_IRQS, which on the CC3220 isn't valid from what I can tell.

ioannisg, 2019-08-23:
@galak Yes, I 've just realized that, and will fix this, so I have moved the milestone back to 2.0 :)

vanti, 2019-08-23:
@galak @ioannisg I think the number of irqs on the cc3220sf is wrong based on the behavior seen when I try to set the Interrupt Set Enable registers. I am submitting #18623 to fix this.

PR #18623 (2019-08-23): soc: arm: ti_simplelink: Fix number of interrupt lines on CC3220SF
There are only 178 interrupt lines on CC3220SF. Hence we should not set
NUM_IRQS to a value exceeding that value. We are changing it to 178.

Fixes #18593
commit ae2742dc (2019-08-23): soc: arm: ti_simplelink: Fix number of interrupt lines on CC3220SF

There are only 178 interrupt lines on CC3220SF. Hence we should not set
NUM_IRQS to a value exceeding that value. We are changing it to 178.

Fixes #18593
PR #18624 (2019-08-25): tests: arm: zero latency irqs: fix the derivation of free NVIC line
The commit fixes the way we determine an available
NVIC IRQ line to perform the zero-latency IRQ test.
The test can now run properly on SOCs that do not
have a continous set of implemented NVIC IRQ lines.

Fixes #18593
closed 2019-08-23 by ioannisg