zephyrproject-rtos/zephyr #18595: USB Console and Bluetooth Samples

bfrog, 2019-08-22:
**Describe the bug**
I'm looking to try the peripheral_hr sample using usb console. So I added the prj.conf settings from the subsys/usb/console/prj.conf to the peripheral_hr/prj.conf and attempted to run it.

This results in the following message from linux
```
[24509.564181] usb 2-9.3.2: new full-speed USB device number 16 using xhci_hcd
[24509.663820] usb 2-9.3.2: New USB device found, idVendor=2fe3, idProduct=0004, bcdDevice= 0.00
[24509.663823] usb 2-9.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[24509.663825] usb 2-9.3.2: Product: Zephyr USB console sample
[24509.663826] usb 2-9.3.2: Manufacturer: ZEPHYR
[24509.663827] usb 2-9.3.2: SerialNumber: 0.01
[24509.671949] cdc_acm 2-9.3.2:1.0: ttyACM0: USB ACM device
[24692.396706] usb 2-9.3.2: USB disconnect, device number 16
[24692.398802] cdc_acm 2-9.3.2:1.0: failed to set dtr/rts
```
When I go to debug I see this loop never exits

```
#if defined(CONFIG_USB_UART_CONSOLE) && defined(CONFIG_USB_UART_DTR_WAIT)
599        while (1) {
600            u32_t dtr = 0U;
601    
602            uart_line_ctrl_get(uart_console_dev, LINE_CTRL_DTR, &dtr);
603            if (dtr) {
604                break;
605            }
606        }
```

**To Reproduce**
Add prj.conf settings from the usb console sample to the peripheral_hr sample, try to run.

**Expected behavior**
I expected usb and bluetooth to be working.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr 0.10.2
 - Commit SHA 9346ded0af6b95d11d6b54e1cbc1f25367888b51

**Additional context**
nrf52840 pca10056 based custom board.

carlescufi, 2019-08-23:
@bfrog can you try https://github.com/zephyrproject-rtos/zephyr/pull/18573 ?

carlescufi, 2019-08-26:
@bfrog it's merged now, so you can test the latest master

jfischer-phytec-iot, 2019-08-27:
quickly tested on reel board with configuration below, seems to work, IMHO configuration problem

```
CONFIG_BT=y                                                                                   
CONFIG_BT_DEBUG_LOG=y                                                                         
CONFIG_BT_SMP=y                                                                               
CONFIG_BT_PERIPHERAL=y                                                                        
CONFIG_BT_GATT_DIS=y                                                                          
CONFIG_BT_GATT_DIS_PNP=n                                                                      
CONFIG_BT_GATT_BAS=y                                                                          
CONFIG_BT_GATT_HRS=y                                                                          
CONFIG_BT_DEVICE_NAME="Zephyr Heartrate Sensor"                                               
CONFIG_BT_DEVICE_APPEARANCE=833                                                               
                                                                                              
CONFIG_GPIO=y                                                                                 
                                                                                              
CONFIG_USB=y                                                                                  
CONFIG_USB_DEVICE_STACK=y                                                                     
CONFIG_USB_DEVICE_PRODUCT="Zephyr USB console sample"                                         
CONFIG_USB_UART_CONSOLE=y                                                                     
                                                                                              
CONFIG_UART_INTERRUPT_DRIVEN=y                                                                
CONFIG_UART_LINE_CTRL=y                                                                       
CONFIG_UART_CONSOLE_ON_DEV_NAME="CDC_ACM_0"                                                   
                                                                                              
CONFIG_USB_UART_DTR_WAIT=n                  
```

```
[00:00:00.000,976] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.000,976] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 0
[00:00:00.001,708] <inf> bt_hci_core: Identity: c8:e9:b4:3c:df:40 (random)
[00:00:00.001,739] <inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 01
[00:00:00.001,739] <inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff
[00:00:00.001,739] <wrn> usb_cdc_acm: Device not configured, drop 1 bytes
[00:00:00.001,861] <wrn> usb_cdc_acm: Device not configured, drop 1 bytes
[00:00:00.005,798] <inf> usb_cdc_acm: USB device suspended
[00:00:00.101,806] <wrn> usb_cdc_acm: Device not configured, drop 1 bytes
[00:00:00.110,351] <inf> usb_cdc_acm: USB device resumed
[00:00:00.110,443] <wrn> usb_cdc_acm: Device not configured, drop 1 bytes
[00:00:00.210,479] <wrn> usb_cdc_acm: Device not configured, drop 1 bytes
[00:00:00.238,067] <wrn> usb_cdc_acm: Device not configured, drop 1 bytes
[00:00:00.290,863] <wrn> usb_cdc_acm: Device not configured, drop 1 bytes
[00:00:00.338,134] <wrn> usb_cdc_acm: Device not configured, drop 1 bytes
[00:00:00.389,099] <inf> usb_cdc_acm: USB device configured
```

jfischer-phytec-iot, 2019-08-27:
@bfrog 
> When I go to debug I see this loop never exits

Please use CONFIG_USB_UART_DTR_WAIT=n or press a key in your terminal client.