zephyrproject-rtos/zephyr #18608: PCA9685 PWM driver is broken

tbursztyka, 2019-08-23:
It misses the API function get_cycles_per_sec() which is mandatory to get pwm.h public function to work.

So currently this driver just does not work at all due to that.

galak, 2019-08-23:
@tbursztyka I think this driver was added for an x86 platform that we have since removed.  Are you wanting to use this on some other board?  Curious how you are hitting this.

tbursztyka, 2020-01-09:
@galak PCA9685 is an i2c slave device. There is no board using it yet, but what about zephyr user that would wire such device on a board? 
Hitting it is easy: just reading the driver code.

jfischer-phytec-iot, 2020-01-09:
There are expansion boards with PCA9685, for example https://www.adafruit.com/product/815

vagotec, 2020-11-07:
Hi,
is there any example code for https://www.adafruit.com/product/815?

Thanks

sarvg

nixward, 2020-11-08:
> Hi,
> is there any example code for https://www.adafruit.com/product/815?
> 
> Thanks
> 
> sarvg
Hi @vagotec if an [overlay](https://docs.zephyrproject.org/latest/reference/devicetree/index.html) was written for the PCA9685 the following sample could be used:
https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/drivers/led_pwm

closed 2021-01-19 by nashif