zephyrproject-rtos/zephyr #18615: sam e70 xplained failed to build hello world

JonathanWoodward, 2019-08-23:
**Describe the bug**
When building same70q21.h is not found

**To Reproduce**
Steps to reproduce the behavior:
1. cd $ZEPHYR_BASE/samples/hello_world
2. mkdir build && cd build
3. cmake -GNinja -DBOARD=sam_e70_xplained ..
4. nijna

**Impact**
showstopper

**Screenshots or console output**
[6/110] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj 
ccache /home/user/LinuxDevelopment/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=v2.0.0-rc1-181-gfbcbc2ed1626 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I../../../kernel/include -I../../../arch/arm/include -I../../../include -I../../../include/drivers -Izephyr/include/generated -I../../../soc/arm/atmel_sam/same70 -I../../../soc/arm/atmel_sam/common/. -I../../../ext/hal/cmsis/Include -isystem ../../../lib/libc/minimal/include -isystem /home/user/LinuxDevelopment/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /home/user/LinuxDevelopment/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Os -imacros/home/user/LinuxDevelopment/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m7 -imacros/home/user/LinuxDevelopment/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/user/LinuxDevelopment/zephyr/samples/hello_world=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/user/LinuxDevelopment/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mabi=aapcs -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj   -c /home/user/LinuxDevelopment/zephyr/arch/arm/core/offsets/offsets.c
In file included from ../../../include/arch/arm/cortex_m/cmsis.h:17,
                 from ../../../include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h:10,
                 from ../../../include/arch/arm/cortex_m/mpu/arm_mpu.h:13,
                 from ../../../include/arch/arm/arch.h:246,
                 from ../../../include/arch/cpu.h:17,
                 from ../../../include/kernel_includes.h:34,
                 from ../../../include/kernel.h:17,
                 from ../../../kernel/include/kernel_structs.h:10,
                 from /home/user/LinuxDevelopment/zephyr/arch/arm/core/offsets/offsets.c:26:
../../../soc/arm/atmel_sam/same70/soc.h:39:10: fatal error: same70q21.h: No such file or directory
 #include <same70q21.h>
          ^~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.

**Environment (please complete the following information):**
 - OS: Linux ubuntu 5.0.0-25-generic #26-Ubuntu SMP Thu Aug 1 12:04:58 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
 - Toolchain zephyr-sdk-0.10.3
 - Commit SHA or Version used https://github.com/zephyrproject-rtos/zephyr.git

galak, 2019-08-23:
Did you do a `west update` ?  I'm guessing you don't have the HAL module now that we split things out.

I'm able to build this w/o any issue.

JonathanWoodward, 2019-08-23:
Hi @galak 

I followed these instructions: 
https://docs.zephyrproject.org/1.11.0/getting_started/getting_started.html
and
https://docs.zephyrproject.org/1.11.0/boards/arm/sam_e70_xplained/doc/sam_e70_xplained.html

There is nothing on here for west??

galak, 2019-08-23:
Looks like you are looking at a pretty old version of the docs (For release 1.11.0)

Try:
https://docs.zephyrproject.org/latest/getting_started/index.html

JonathanWoodward, 2019-08-23:
@galak I tried installing west as shown

$ pip3 show -f west
Name: west
Version: 0.6.0
Summary: Zephyr RTOS Project meta-tool
Home-page: https://github.com/zephyrproject-rtos/west
Author: Zephyr Project
Author-email: devel@lists.zephyrproject.org
License: UNKNOWN
Location: /home/user/.local/lib/python3.7/site-packages
Requires: setuptools, PyYAML, configobj, colorama, pykwalify
Required-by: 
Files:
  ../../../bin/west
  west-0.6.0.dist-info/DESCRIPTION.rst
  west-0.6.0.dist-info/INSTALLER
  west-0.6.0.dist-info/METADATA
  west-0.6.0.dist-info/RECORD
  west-0.6.0.dist-info/WHEEL
  west-0.6.0.dist-info/entry_points.txt
  west-0.6.0.dist-info/metadata.json
  west-0.6.0.dist-info/top_level.txt
  west/__pycache__/backports.cpython-37.pyc
  west/__pycache__/build.cpython-37.pyc
  west/__pycache__/cmake.cpython-37.pyc
  west/__pycache__/configuration.cpython-37.pyc
  west/__pycache__/log.cpython-37.pyc
  west/__pycache__/main.cpython-37.pyc
  west/__pycache__/manifest.cpython-37.pyc
  west/__pycache__/util.cpython-37.pyc
  west/__pycache__/version.cpython-37.pyc
  west/backports.py
  west/build.py
  west/cmake.py
  west/commands/__init__.py
  west/commands/__pycache__/__init__.cpython-37.pyc
  west/commands/__pycache__/config.cpython-37.pyc
  west/commands/__pycache__/project.cpython-37.pyc
  west/commands/config.py
  west/commands/project.py
  west/commands/west-commands-schema.yml
  west/configuration.py
  west/log.py
  west/main.py
  west/manifest-schema.yml
  west/manifest.py
  west/util.py
  west/version.py

However the west command and ../../../bin/west does not exist

galak, 2019-08-23:
Can you try the following:

```
[galak@lava ~]$ which west
~/.local/bin/west
[galak@lava ~]$ echo $PATH
```

Wondering if `west` isn't in your path.

JonathanWoodward, 2019-08-23:
user@ubuntu:~$ which west
user@ubuntu:~$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
user@ubuntu:~$

galak, 2019-08-23:
Yep, you need `~/.local/bin` in your PATH.

JonathanWoodward, 2019-08-23:
That fixed it. It sets the PATH on mac but not on Linux.

carlescufi, 2019-08-23:
@JonathanWoodward note that this is described in the very beginning of the Getting Started Guide:
https://docs.zephyrproject.org/latest/getting_started/index.html#python-and-pip

JonathanWoodward, 2019-08-23:
zephyrproject/zephyr$ west build -b sam_e70_xplained samples/hello_world
-- west build: build configuration:
       source directory: /home/user/LinuxDevelopment/zephyrproject/zephyr/samples/hello_world
       build directory: /home/user/LinuxDevelopment/zephyrproject/zephyr/build
       BOARD: sam_e70_xplained (origin: CMakeCache.txt)
-- west build: building application
Error: could not find CMAKE_PROJECT_NAME in Cache
ERROR: command exited with status 1: /usr/bin/cmake --build /home/user/LinuxDevelopment/zephyrproject/zephyr/build

galak, 2019-08-23:
> zephyrproject/zephyr$ west build -b sam_e70_xplained samples/hello_world
> -- west build: build configuration:
> source directory: /home/user/LinuxDevelopment/zephyrproject/zephyr/samples/hello_world [...]

Try removing the build dir and than trying it.

`rm -fr /home/user/LinuxDevelopment/zephyrproject/zephyr/build`

Also if you're able to join slack chat, might be easier to get interactive help (see Chat) here:

https://docs.zephyrproject.org/latest/introduction/index.html#resources

JonathanWoodward, 2019-08-23:
I would but I do not have @intel.com, @linaro.com, @linux.intel.com, @nordicsemi.no, @nxp.com, @oticon.com, @foundries.io, @ti.com, @synopsys.com, @antmicor.com, or @sifive.com email address

carlescufi, 2019-08-23:
Closing this issue, please contact us on Slack or open a new one if you continue to have bulid issues

closed 2019-08-23 by carlescufi

carlescufi, 2019-08-23:
@JonathanWoodward use this link: https://tinyurl.com/y5glwylp

carlescufi, 2019-08-23:
> zephyrproject/zephyr$ west build -b sam_e70_xplained samples/hello_world
> -- west build: build configuration:
> source directory: /home/user/LinuxDevelopment/zephyrproject/zephyr/samples/hello_world [...]

Delete the build folder and try again

carlescufi, 2019-08-23:
> I would but I do not have @intel.com, @linaro.com, @linux.intel.com, @nordicsemi.no, @nxp.com, @oticon.com, @foundries.io, @ti.com, @synopsys.com, @antmicor.com, or @sifive.com email address

You are using an old Slack link, use the one I posted above