zephyrproject-rtos/zephyr #18652: Optimization flags from CMAKE_BUILD_TYPE are not taken into account

meuter, 2019-08-26:
**Describe the bug**
When compiling using cmake you can usually provide `-DCMAKE_BUILD_TYPE=Release` (or `Debug`, or `MinRelSize`) and depending on the build type, the optimization level will be adapted (`-O3` for Release, `-O0` for `Debug` and `-Os` for `MinRelSize`), among other things. But when building a zephyr app, these flags are not taken into account. Apparently, an extra optimization flag is added to the compiler flags from the KConfig option:

"Build and Link Features → Compiler Options → Optimization level"

which by default is "Optimize for size" (`-Os`). And since this flags is added to the CFLAGS after the flags coming from CMAKE_BUILD_TYPE, it overrides it.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/hello_world/; mkdir build; cd build
2. export BOARD=qemu_x86 # does not matter
3. cmake ../ -G"Unix Makefiles" -DCMAKE_BUILD_TYPE=Release
4. make VERBOSE=1
5. check the compiler flags and you'll see ` -O3 -DNDEBUG`  (from CMAKE_BUILD_TYPE) and a little later  `-Os` from the KConfig.

**Expected behavior**
I would expect the flags from CMAKE_BUILD_TYPE to be taken into account instead of the one for KConfig as it's standard behavior for any cmake project. 

Another option, if KConfig is the preferred way, would be to have cmake detect that and error out mentioning that CMAKE_BUILD_TYPE is not supported or something along those lines.

**Environment (please complete the following information):**
 - Linux, Ubuntu 16.10
 - Zephyr SDK 0.10.3
 - Zephyr 1.14.0
 - cmake  3.13.1

SebastianBoe, 2019-08-26:
Zephyr used Kconfig before CMake and therefore uses Kconfig to set optimization flags.

AFAICT using Kconfig to set these flags works just fine so this is not going to change.

I'm not sure how to communicate that CMAKE_BUILD_TYPE will not change optimization flags. I'd prefer not to error out when CMAKE_BUILD_TYPE is set, because app-developers may want to use it to change how the app is built.

Perhaps we could at least unset the CMAKE_C_FLAGS_[DEBUG|RELEASE|RELWITHDEBINFO|MINSIZEREL|...] variables so that we don't have conflicting/overriding flags in the command.

carlescufi, 2019-08-26:
Reclassified from bug to question since this is really not a bug, it's intentional and configurable via Kconfig instead of CMake.

meuter, 2019-08-26:
@carlescufi fair enough ^_^

@SebastianBoe I understand that KConfig is the preferred way, but as it stands, it's a footgun IMHO.  The use of CMAKE_BUILD_TYPE is very idiomatic in the cmake community and a lot of people coming from this community will expect this. Erroring out may indeed be a little abrupt, a warning may be more appropriate in this case, and it could point to the KConfig option? 

Another maybe more flexible possibility would be to add a "CMAKE_OPTIMIZATIONS" config in the "Optimization level" choice that would not add any optimization flags. With this people preferring the cmake way of doing things (like myself ^_^) would have an option.

Note that both adding a warning and the addition choice are not necessarily mutually exclusive.

meuter, 2019-08-27:
Just to see what this would look like, I drafted a patch for the extra choice that seems to be working:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index d9626390fb..022bb91021 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -144,6 +144,8 @@ elseif(CONFIG_SPEED_OPTIMIZATIONS)
   set(OPTIMIZATION_FLAG ${OPTIMIZE_FOR_SPEED_FLAG})
 elseif(CONFIG_SIZE_OPTIMIZATIONS)
   set(OPTIMIZATION_FLAG ${OPTIMIZE_FOR_SIZE_FLAG}) # Default in kconfig
+elseif(CONFIG_CMAKE_OPTIMIZATIONS)
+  # Let cmake handle optimization flags using CMAKE_BUILD_TYPE
 else()
   assert(0 "Unreachable code. Expected optimization level to have been chosen. See Kconfig.zephyr")
 endif()
diff --git a/Kconfig.zephyr b/Kconfig.zephyr
index bf34d67f86..abe3817f9e 100644
--- a/Kconfig.zephyr
+++ b/Kconfig.zephyr
@@ -240,6 +240,11 @@ config NO_OPTIMIZATIONS
 	  Compiler optimizations will be set to -O0 independently of other
 	  options.
 
+config CMAKE_OPTIMIZATIONS
+	bool "Use cmake optimization flags"
+	help
+	  Compiler optimizations will be set according to CMAKE_BUILD_TYPE
+
 endchoice
 
 config COMPILER_OPT
diff --git a/samples/hello_world/prj.conf b/samples/hello_world/prj.conf
index b2a4ba5910..2c433b8199 100644
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1 @@
-# nothing here
+CONFIG_CMAKE_OPTIMIZATIONS=y
```

SebastianBoe, 2019-08-27:
Unfortunately, Kconfig needs to know the optimization level to e.g. set buffer sizes appropriately, so a CMAKE_OPTIMIZATIONS option won't work as intended.

meuter, 2019-08-28:
That's indeed a bummer. Well it was worth a try. Then removing the optimization flags from CMAKE_C_FLAGS_<xxx> as you suggested and maybe adding a warning/info message at configure time seems like the best option.

SebastianBoe, 2019-08-29:
I made an attempt at unsetting CMAKE_C_FLAGS_xx. But unfortunately it is set by 'project' invocation's in app build scripts, e.g.

https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/hello_world/CMakeLists.txt#L6

and we don't have any code running after this. So we can't enforce it to be unset.

PR #18777 (2019-11-06): cmake: Check if match between CMAKE_BUILD_TYPE and OPTIMIZATION_FLAG
In CMake projects, 'CMAKE_BUILD_TYPE' usually determines the
optimization flag, but in Zephyr it is determined through Kconfig.

To avoid confusing users we now give a warning if there is a mismatch
between the CMAKE_BUILD_TYPE and the optimization flag.

This fixes #18652
closed 2019-11-06 by galak