zephyrproject-rtos/zephyr #18712: ARM Cortex-M: Z_ARCH_EXCEPT not working for Baseline SOCs

ioannisg, 2019-08-27:
**Describe the bug**
Z_ARCH_EXCEPT() is called, to signal errors directly from Software. In ARM Cortex-M, this is implemented with a synchronous SVC instruction, with code 2.

In Mainline SOCs this works well, because the kernel reserves the highest interrupt priority for the SVC Handler.

In Baseline cores, however, we do not reserve an interrupt priority for the SVC. So if Z_ARCH_EXCEPT() is invoked from an interrupt with the highest priority, the SVC Call won't be able to execute and will trigger a synchronous (Escalated) hard fault.

**To Reproduce**
We have a bug report already for this: #18590 

**Impact**
We are not able to infer stack-fail checks in Baseline ARM CPUs with the STACK Sentinel feature.

We do get a system exception, however, so the user is still getting some information that something went wrong.

ioannisg, 2019-10-04:
There is a work-around for this already merged, hence this should remain a low priority issue.

dleach02, 2019-12-10:
Removing 2.1 milestone. @ioannisg, can you add a reference to the PR of the workaround you referenced?

ioannisg, 2019-12-13:
> Removing 2.1 milestone. @ioannisg, can you add a reference to the PR of the workaround you referenced?

https://github.com/zephyrproject-rtos/zephyr/pull/18713

This PR fixed this by doing some inspection of the escalated hard fault.

ioannisg, 2020-07-21:
Thinking about this, maybe we cab turn this to an enhancement, because the workaround has been stable for quite some time now.
Z_ARCH_EXCEPT _does_ work for Baseline cortex-m via the hard-fault inspection. Need to think if this should actually be made to work as it does for Mainline: it would probably eat an IRQ level for SVC and we don't have many levels available in Baseline.