zephyrproject-rtos/zephyr #18728: tests/subsys/logging/log_core failed on sam_e70 board.

chen-png, 2019-08-28:
**Describe the bug**
Running test suite test_log_list on sam_e70 board, it failed with error message "Unexpected message index".

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=sam_e70_xplained tests/subsys/logging/log_core
2. west flash

**Screenshots or console output**

***** Booting Zephyr OS build v2.0.0-rc1-249-gcac26db2bc69 *****

Running test suite test_log_list

starting test - test_log_backend_runtime_filtering

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:67: put: (log_msg_timestamp_get(msg) not equal to exp_timestamp)
Unexpected message index
FAIL - test_log_backend_runtime_filtering

starting test - test_log_overflow

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:67: put: (log_msg_timestamp_get(msg) not equal to exp_timestamp)
Unexpected message index
FAIL - test_log_overflow

starting test - test_log_arguments

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:303: test_log_arguments: (8 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_log_arguments

starting test - test_log_from_declared_module

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:326: test_log_from_declared_module: (2 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_log_from_declared_module

starting test - test_log_strdup_gc

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:348: test_log_strdup_gc: (2 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_log_strdup_gc

starting test - test_log_strdup_detect_miss
PASS - test_log_strdup_detect_miss

starting test - test_strdup_trimming

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:430: test_strdup_trimming: (1 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_strdup_trimming

starting test - test_log_msg_dropped_notification

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:445: log_n_messages: (backend1_cb.total_drops not equal to exp_dropped)
Unexpected log msg dropped
FAIL - test_log_msg_dropped_notification

starting test - test_log_panic

    Assertion failed at ZEPHYR_BASE/tests/subsys/logging/log_core/src/log_core_test.c:497: test_log_panic: (2 not equal to backend1_cb.counter)
Unexpected amount of messages received by the backend.
FAIL - test_log_panic

Test suite test_log_list failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.10.3
 - Commit ID: cac26db2bc69

pizi-nordic, 2019-08-28:
It looks that this test assumes that it is the sole log messages source, which might be not true in your case. Please disable all other log sources and run the test.

You can check what is enabled using the following command in the build directory:
```
$ grep LOG_LEVEL= zephyr/.config  | grep -v =0
CONFIG_MPU_LOG_LEVEL=3
CONFIG_UART_CONSOLE_LOG_LEVEL=3
CONFIG_WDT_LOG_LEVEL=3
CONFIG_IPM_LOG_LEVEL=3
CONFIG_UPDATEHUB_LOG_LEVEL=3
```

nashif, 2019-08-29:
@chen-png please confirm that you can reproduce this.

ioannisg, 2019-08-29:
Let me try, as well on my sam_e70

ioannisg, 2019-08-29:
The test fails for me on sam_e70_xplained

chen-png, 2019-08-30:
This issue is repeatable in the zephyr test report recently. And I have run the "grep LOG_LEVEL= zephyr/.config  | grep -v =0" command in the build directory, it's the same as follows:
CONFIG_MPU_LOG_LEVEL=3
CONFIG_UART_CONSOLE_LOG_LEVEL=3
CONFIG_WDT_LOG_LEVEL=3
CONFIG_IPM_LOG_LEVEL=3
CONFIG_UPDATEHUB_LOG_LEVEL=3
The test report is running on the server, we don't change any files in the zephyr source tree.

stephanosio, 2019-12-02:
I can also reproduce this on the latest master.

closed 2020-11-03 by github-actions[bot]